openpnp/openpnp #336: nozzleTip  rotation is buggy

cri-s, 2016-08-30:
If rotation is final angle it raises division by zero.
Suppose calibration inter
It calc wron

cri-s, 2016-08-30:
Actually as note to myself
Severity: high.
Suppose calibration interval 15 deg.
Possible div zero for ratio , not bad if result is not NaN or exception.
If angle is >= 345 deg calc is wrong.
Nozzle moves in x and y during calibration, this is bad and introduce movement/backlash error, specially if rails are a bit sticky.
Nozzle Tip should be calibrated only on each change if nozzleTip change is made manually.
Nozzle position center is not used and as result calibration has offsets.
As note, homing must ensure cam position is near logical Zero to minimize thermal expansion error.
As note, if up looking cam is refocused or moved head 0 must be recalibrated.

vonnieda, 2016-08-30:
Please add more detail to this issue. It is not clear what feature or code you are referring to, or what the problem is. See: https://github.com/openpnp/openpnp/blob/develop/CONTRIBUTING.md#issues-bug-reports-and-feature-requests

cri-s, 2016-08-30:
If angle >= 345 when using 15 deg calib intervals the resulting offset is wrong.
Reason is, ratio is calc wrong because of assumption offsetB is always higher as offsetA fails.

On calibration, nozzle is moved in order nozzle is centered over camera dropping this offset. That's wrong and create artificial offset. 

How to reproduce that. Remove nozzle, place colored sticker on one side of nozzle holder and program vision to find that color. This create offset of some mm. Now you can test it and see how off is it in reality.

cri-s, 2016-08-30:
The above is the two major issues, two minor are described

vonnieda, 2016-08-30:
@cri-s So, this is all related to nozzle tip calibration, correct? I want to correct the title. "nozzleTip rotation is buggy" does not imply nozzle tip calibration.

cri-s, 2016-08-30:
Yes, my fault.

vonnieda, 2016-10-28:
Related to #235.

closed 2016-10-28 by vonnieda