openpnp/openpnp #337: nozzleTip  calibration result changes position.

cri-s, 2016-08-30:
nozzleTip  calibration result changes position on dro,
Break all tools that get position from nozzle and breaks nozzles>camera button.
Severity: high
Workaround none
Fix: nozzle need to have persistent  nozzleTipOffset location and getHeadOffsets need adding that value.
Notes: nozzleTip load and unload code need clear nozzleTipOffset or use override function.

vonnieda, 2016-08-30:
This work is being done in #235. Closing this one as duplicate, but adding a note there to check these comments to fix.

closed 2016-08-30 by vonnieda

cri-s, 2016-08-30:
The issue here is that it changes the dro x and y position breaking all
functions that from nozzle gets position and transfer it as pickup position
or use this as camera position.
If nozzle is 1mm off, then correction code position nozzle 1mm away from
programmed position. That is correct but then if using down looking camera,
it gets the position from nozzle, that are 1mm wrong, and vision then
return 1mm offsets, that nozzle use and fails to pick 0603 component. Are
you sure it's duplicated issue?

vonnieda, 2016-08-30:
@cri-s That's fine. These comments can be looked at when we get to finishing the nozzle tip calibration. The nozzle tip calibration code is known to not be working, so it needs more work before we can use it anyway. I'll make sure to look at these comments (and have linked them in the other issue) before finishing it. In the mean time, I am going to remove the feature so it stops causing confusion.