openpnp/openpnp #338: future (threading) abstraction makes timing failure.

cri-s, 2016-08-30:


cri-s, 2016-08-30:
Severity: low.
Description:
Its normal that GUI ui have its own thread, but main thread need to be in sync  at least with motion and camera, actuators and other interfaces, otherwise delay need have extra time. As example its a fault doing camera motion using future interface and setting timer to take picture after expiration. Doing it so its not granted that camera is out of motion. Example of error source: USB reconnect.
Using futures is possible, I'm not saying is wrong and I could be wrong with my teardown of error analysis.

vonnieda, 2016-08-30:
Please add more detail to this issue. It is not clear what feature or code you are referring to, or what the problem is. See: https://github.com/openpnp/openpnp/blob/develop/CONTRIBUTING.md#issues-bug-reports-and-feature-requests

If this is a feature request, please specify exactly what you are looking for. If it a bug report, please include step by instructions on how to reproduce the error.

cri-s, 2016-08-30:
Too hard to make test case.

closed 2016-08-30 by cri-s