openpnp/openpnp #339: homing dot fails.

cri-s, 2016-08-30:
Severity high
Description: bad implementation.
It don't repeat fiducial 3 times, it discard results.
It should be moved to head as it was already implemented there. Need to check nozzle offsets to and raise error if wrong.

vonnieda, 2016-08-30:
Please add more detail to this issue. See: https://github.com/openpnp/openpnp/blob/develop/CONTRIBUTING.md#issues-bug-reports-and-feature-requests

We need step by step instructions on how to reproduce the error, and a description of what the error is.

mattbrockuk, 2016-08-30:
hi cri

is this the homing fiducial code i added?

thanks

matt

cri-s, 2016-08-30:
Yes
How to check: make homing dot 6mm from 0 position.
Check it works.
Now feeder or every other fixed configured position should be 6mm off, but surprise surprise even homing dot was found it's ignored.

cri-s, 2016-08-30:
Another thing, homing dot, the most important fiducial is checked only 1 time. Other fiducials are checked 3 times moving over it in order to minimize camera distortion errors and motion errors.

vonnieda, 2016-08-30:
Related to #274

mattbrockuk, 2016-08-30:
Hi Cri

Ok i'll look at this... 

PS> i'm not sure if its a translation thing... but you come across as quite arrogant/angry?

Matt

mattbrockuk, 2016-08-30:
I've had a think and i've realised whats wrong...

Absolute vs relative gcode positioning

Currently the code finds the fiducial and it zeros the openpnp axis at this point, this is home. This works if using relative positioning - openpnp will go - move 100mm left - and as its had its home set at the fiducial - this will account for any homing error.

What it won't work for is absolute positioning, openpnp will home and zero its axis at the fiducial, but then it will go - goto 10mm,10mm - to the driver (whose not had its axis zero'd) and the driver will just go to 10mm,10mm regardless of where openpnps home position is. What we need to do is at the offset found at the fiducial home to the absolute position. So say it homes and our home fiducial is +3mm away, if it wants to go to position 10mm,10mm (say a feeder position), it should be commanded to go to 13mm,10mm to take into account the offset.

I think this makes sense?

cri-s, 2016-08-31:
Il 31/ago/2016 00:48 "mattbrockuk" <notifications@github.com> ha scritto:
>
> I've had a think and i've realised whats wrong...
>
> Absolute vs relative gcode positioning
>

>

You must send the coordinate to Gcode driver.
As example if absolute position should be
Zero,
Then gcode need be at minimum.
G92 X0 Y0
Ideally it should be build from homeCoordinate
and Axis name. Don't forget to use US locals
in the case of building the G92 command.

You can include other axes as well, as example for setting Z to different
height or inverting sign and x and y could be not 0.
Eventually it can be opportune to let user define fiducial location
different as zero.

Another weak point is that fiducials are searched only one time. It should
do 3 times in case of uncalibrated lenses with a lot of distortion.
Just call the function 3 .
.

vonnieda, 2016-08-31:
We can't hard code G92. We can define it as a command like the rest of the
Gcodes if sending G92 is the best option. I'd prefer, though, if we just
handle the offsets internally if possible. I think some thought should be
given to how a user will actually use this and whether this is something we
can offload to the controller using the home coordinates.

Jason

vonnieda, 2016-08-31:
To expand on my comment above: All that really needs to happen after homing is that the controller and the driver both need to be on the same page as to what the current coordinates are. Perhaps we could do this by adding a new command: POST_VISION_HOME_COMMAND. Users can set this command to, for instance, G92 X0 Y0, or whatever suits their controller.

cri-s, 2016-08-31:
G92 is standard gcode and all controller support it.
If not homing is not usable.
It's possible to have variable offset and if its not null,
Then it's added to every moveTo location argument.

Another possibility is to add offset variable to head,
and do homing dot inside ReferenceHead modify getHeadOffset to include
getHeadOffset from head.
The real problems with that is if having 2 head and only one
camera. Then it fails.

vonnieda, 2016-08-31:
@cri-s It's standard for Gcode controllers, but people are starting to use this driver for non-Gcode controllers and I want to support that use case. There is no downside in making it a configurable command.

cri-s, 2016-08-31:
Then local variable offset should be used.

  Location homeOffset; // inside class

Inside home:

  homeOffset = Configuration.get().getMachine().getFiducialLocator()
                    .getHomeFiducialLocation(tmp, homePart);

 if(homeOffset!=null) {
homeOffset=Configuration.get().getMachine().getFiducialLocator()

                    .getHomeFiducialLocation(tmp, homePart);

homeOffset=Configuration.get().getMachine().getFiducialLocator()

                    .getHomeFiducialLocation(tmp, homePart);
}

Inside move to:

if (home offset!= null) { location = location.add(home offset); }

vonnieda, 2016-08-31:
I think that would be a good solution too. I'm fine with either way.

mattbrockuk, 2016-08-31:
Ok i'll look at doing that thanks

tracklogic, 2016-08-31:
i'm thinking something as easy as

           String g92command = getCommand(null, CommandType.POST_VISION_HOME_COMMAND);
            g92command = substituteVariable(g92command, "X", 0);
            g92command = substituteVariable(g92command, "Y", 0);
            sendGcode(g92command, -1);

tracklogic, 2016-08-31:
Well actually... as its the camera that will be above the fiducial, we should set it to the camera offset?

 String g92command = getCommand(null, CommandType.POST_VISION_HOME_COMMAND);
            g92command = substituteVariable(g92command, "X", camera.getLocation().getX());
            g92command = substituteVariable(g92command, "Y", camera.getLocation().getY());
            sendGcode(g92command, -1);

@vonnieda

vonnieda, 2016-09-01:
@tracklogic I think you'd want to set it to the home coordinate for each axis, defined on the axis.

tracklogic, 2016-09-07:
Ive put a pull request in for this

vonnieda, 2016-09-09:
Resolved in #345

closed 2016-09-09 by vonnieda