openpnp/openpnp #343: Actuator UI Improvements

vonnieda, 2016-09-04:
Discussion in: https://groups.google.com/d/msgid/openpnp/CA%2BQw0jwW%3D8dO5jNV37mip0Lp9OeWa2LkAw8sh-JCYM0ALw4hfQ%40mail.gmail.com?utm_medium=email&utm_source=footer

* Need a way to trigger both boolean and double actuators.
* Need to be able to see the state of the boolean actuator better, or alternately, just be able to send a fixed true or false.

michalsok, 2017-09-26:
I am new to OpenPNP but I have noticed that Actuators are named and have an ID so there is an index / array of them of some sort. (or could easily be populated) 
Why not make the list of defined actuators available as a drop-down selector in places where they are selected like drag feeders, auto-feeders etc. Right now you have to type in the actuator name.
Thank you,

vonnieda, 2017-09-26:
@michalsok Good idea!

markmaker, 2021-04-25:
Some of this seems to already have been done in the meant-time:

Actuators are now drop-downs everywhere, as requested in the comment here. 

From the linked discussion, Jason said:
> Right now there's no way to tell if an actuator takes a boolean or double argument. It may make sense to add a flag that says which is which, and then we can have the UI react to that: checkbox for boolean actuators and a textfield for entering a value for double ones. 

#1095 added actuators types and fulfills this. 

[image]

In addition, **Default ON** and **Default OFF** values that can now be defined for non-boolean actuators for quick switching between two predefined states. Profile type actuators can provide selectable options.

closed 2021-04-25 by vonnieda