openpnp/openpnp #349: Camera to board Error

psykhon, 2016-10-22:
The "position the camera at the board's current location" button is now actuating the noozle, that didnt happen before

vonnieda, 2016-10-22:
@psykhon Can you describe the problem step by step? I just tested this with current code and I don't see any issue. It may be that your head offsets or Gcode axis mappings are incorrect.

psykhon, 2016-10-22:
Its also happening when I hit the "perform a fiducial check button..." , it lowers the noozle on each fiducial, lifting it between

my machine.xml
`
<openpnp-machine>
   <machine class="org.openpnp.machine.reference.ReferenceMachine" speed="5.0">
      <heads>
         <head class="org.openpnp.machine.reference.ReferenceHead" id="22964dce-252a-453e-8106-65db104a0763" name="H1">
            <nozzles>
               <nozzle class="org.openpnp.machine.reference.ReferenceNozzle" id="69edd567-df6c-495a-9b30-2fcbf5c9742f" name="N1" pick-dwell-milliseconds="0" place-dwell-milliseconds="0" current-nozzle-tip-id="e092921a-2eef-449b-b340-aa3f40d8d791" changer-enabled="false" limit-rotation="true">
                  <nozzle-tips>
                     <nozzle-tip class="org.openpnp.machine.reference.ReferenceNozzleTip" id="e092921a-2eef-449b-b340-aa3f40d8d791" name="NT1" allow-incompatible-packages="true">
                        <compatible-package-ids class="java.util.HashSet"/>
                        <changer-start-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-mid-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <changer-end-location units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                        <calibration angle-increment="15.0" >
                           <pipeline>
                              <stages>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.ImageCapture" name="9" enabled="true" settle-first="true"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.ConvertColor" name="1" enabled="true" conversion="Bgr2Gray"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.BlurMedian" name="4" enabled="true" kernel-size="3"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="3" enabled="true" diameter="200"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.Threshold" name="2" enabled="true" threshold="190" auto="true" invert="false"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.MaskCircle" name="5" enabled="true" diameter="100"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.SimpleBlobDetector" name="result" enabled="true"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.ImageRecall" name="0" enabled="true" image-stage-name="9"/>
                                 <cv-stage class="org.openpnp.vision.pipeline.stages.DrawKeyPoints" name="7" enabled="true" key-points-stage-name="result">
                                    <color r="255" g="0" b="204" a="255"/>
                                 </cv-stage>
                              </stages>
                           </pipeline>
                        </calibration>
                     </nozzle-tip>
                  </nozzle-tips>
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </nozzle>
            </nozzles>
            <actuators>
               <actuator class="org.openpnp.machine.reference.ReferenceActuator" id="5b8a8cbd-d6c1-4324-af8b-eba5f0444622" name="A1" index="0">
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </actuator>
               <actuator class="org.openpnp.machine.reference.ReferenceActuator" id="5b8a8cbd-d6c1-4324-af8b-eba5f0444622" name="A2" index="1">
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </actuator>
            </actuators>
            <cameras>
               <camera class="org.openpnp.machine.reference.camera.OpenCvCamera" id="2e78436e-f335-46a3-b099-1e16420c99f6" name="Micro" looking="Down" settle-time-ms="300" rotation="0.0" flip-x="false" flip-y="false" offset-x="0" offset-y="0" crop-width="0" crop-height="0" device-index="0" preferred-width="0" preferred-height="0" fps="24">
                  <units-per-pixel units="Millimeters" x="0.032" y="0.032" z="0.0" rotation="0.0"/>
                  <vision-provider class="org.openpnp.machine.reference.vision.OpenCvVisionProvider"/>
                  <head-offsets units="Millimeters" x="-74.407" y="6.7" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
                  <calibration enabled="false">
                     <camera-matrix length="9">0.0, 0.0, 0.0, 0.0, 0.0, 0.0, 2.49488167E-315, 2.494882027E-315, 9.486E-321</camera-matrix>
                     <distortion-coefficients length="5">0.0, 0.0, 0.0, 0.0, 0.0</distortion-coefficients>
                  </calibration>
               </camera>
            </cameras>
            <paste-dispensers>
               <paste-dispenser class="org.openpnp.machine.reference.ReferencePasteDispenser" id="53050ccf-59a0-4d9f-a8d3-6216f5412e4e" name="D1">
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
                  <safe-z value="0.0" units="Millimeters"/>
               </paste-dispenser>
            </paste-dispensers>
[... cut ...]
                        <cv-stage class="org.openpnp.vision.pipeline.stages.DrawRotatedRects" name="8" enabled="true" rotated-rects-stage-name="result" thickness="2"/>
                        <cv-stage class="org.openpnp.vision.pipeline.stages.ImageWriteDebug" name="15" enabled="true" prefix="bv_result_" suffix=".png"/>
                     </stages>
                  </pipeline>
               </part-settings>
            </entry>
         </part-settings-by-part-id>
      </part-alignment>
      <fiducial-locator class="org.openpnp.machine.reference.vision.ReferenceFiducialLocator"/>
   </machine>
</openpnp-machine>
`

vonnieda, 2016-10-23:
Find your Z axis section, and change it to look like this:

```
                <axis home-coordinate="0.0" name="z" type="Z">
                    <head-mountable-ids class="java.util.HashSet">
                        <string>69edd567-df6c-495a-9b30-2fcbf5c9742f</string>
                    </head-mountable-ids>
                </axis>
```

Note, the only change is to change the * to 69edd567-df6c-495a-9b30-2fcbf5c9742f. This tells OpenPnP to only consider the nozzle (and not the camera, or anything else) for Z movements.

psykhon, 2016-10-23:
Txs! that did it

closed 2016-10-23 by psykhon

vonnieda, 2016-10-25:
Can you describe the problem step by step? I just tested this with current
code and I don't see any issue. It may be that your head offsets or Gcode
axis mappings are incorrect.