openpnp/openpnp #354: Nozzle Tip Changer Should Support 4+ Moves

vonnieda, 2016-10-28:
Certain nozzle tip changer designs require 4 moves, and some may require more. As a first step we should just add a fourth move to the system so that this type of changer can be supported. If it turns out we need more we can switch to an arbitrary list of moves type system later.

Discussion at https://groups.google.com/d/msgid/openpnp/f93df167-9e0d-44c6-892a-f8fd2701eb54%40googlegroups.com?utm_medium=email&utm_source=footer

vonnieda, 2016-10-28:
Note: Per @cri-s we should move to safe-z before the first move and after the last. See https://groups.google.com/d/msgid/openpnp/53e47ee2-f975-4170-b357-96ca4f961f17%40googlegroups.com

Related, I seem to remember changing the code NOT to do this at some point because of an issue someone filed related to testing the movements, but I think that might have been in the location buttons panel. Need to try to find that change and make sure they are not incompatible.

cri-s, 2016-10-28:
> msgid/openpnp/53e47ee2-f975-4170-b357-96ca4f961f17%40googlegroups.com

The requests to not go to safeZ was for manual moving to that position. The
code should only go to safeZ height and not go to final position, and only
on automatic load/unload.
The nozzleChangerPos don't should be configured on gui,
It's optional and on GUI NaN don't work too.

vonnieda, 2016-10-28:
@cri-s What do you mean by "The nozzleChangerPos don't should be configured on gui,
It's optional and on GUI NaN don't work too."? Which position do you mean?

vonnieda, 2016-10-28:
@cri-s Nevermind previous comment, I see you have added a nozzleChangerPos field on Nozzle. I will go back through earlier messages to understand what this is used for.

cri-s, 2016-10-28:
It is used as entry and exit point to travel to nozzle holder.
This gives one the possibility to have defined travel path to nozzle, in
combination with nozzleChangerEndPos in order it don't travel over nozzle
holder if it come from unknown position. As GUI don't support NaN value,
but XML support it,
I would suggest to configure  nozzleChangerPos only manually on XML file
for advanced users and give it default location of NaN. Further actually
the additional initial/final MovableUtils.moveToLocationAtSafeZ(this,
nt.getChangerEndLocation()); is a bug.
It should move to that location at safeZ height and stay at that height
ignoring the Z value.
During tested the wrong move caused nozzle bending and instead of changing
code coordinates was changed in order to avoid this bug.

vonnieda, 2016-10-28:
@cri-s I think I understand. So, this is to handle a case where the nozzle holders are high enough that the nozzle cannot pass cleanly over it?

closed 2016-10-28 by vonnieda

cri-s, 2016-10-28:
Yes.

@cri-s <https://github.com/cri-s> I think I understand. So, this is to
handle a case where the nozzle holders are high enough that the nozzle
cannot pass cleanly over it?

—
<https://github.com/openpnp/openpnp/issues/354#issuecomment-257043193>, or mute
the thread

commit 0704e400 (2016-10-28): Fixes #354, adds a fourth move to the nozzle tip changer. The additional move is placed in the middle and for backwards compatibilty it will be cloned from the previous middle move upon first config load.
commit 01d22e0c (2016-10-28): Merge branch 'develop' into feature/camera-lighting

* develop:
  Updated CHANGES.md
  Fixes #354, adds a fourth move to the nozzle tip changer. The additional move is placed in the middle and for backwards compatibilty it will be cloned from the previous middle move upon first config load.
commit 880019ef (2017-01-07): Merge branch 'develop' into feature/loose-part-feeder

* develop: (96 commits)
  Untitled machine setup tabs filled
  moving generic code for feeders to base class
  moved generic code for cameras to base class ReferenceCamera
  Feeders configuration tab in "Machine Setup" panel no titled with "Configuration"
  Configuration tabs within "Machine Setup" panel now with title
  Added further improvement to startup message by adding 'Namste' along with multi language 'Hello'
  Feeder panel layout in line with other panels
  executing "first pull" tutorial
  Improves error messages for fidicual det [... cut]