openpnp/openpnp #355: Camera Light Control During Vision Operations

vonnieda, 2016-10-28:
There is a need to be able to turn on and off certain lights when performing vision operations. The biggest issue so far is that often the LEDs on the top camera will interfere with the bottom camera when doing bottom vision.

This a feature that could be fairly large, so I am breaking this into what can be done quickly (MVP) and what we can do in the future to make this very flexible (Backlog).

MVP:
- [ ] Add an actuatorName property to ReferenceBottomVision.
- [ ] ReferenceBottomVision should turn on the actuator before performing vision and turn it off when finished. Note that the actuator could be configured (via GcodeDriver, for instance) to turn on the bottom lights and turn off the top lights when on and then vice versa when off.

Backlog:
The Camera interface and implementations need to be reworked so that capture() is only called for vision operations, actuators can be added to cameras for lighting and the actuator will be called for capture. This will make it so that lights are only on when the camera needs them to be.

This requires some thought, though. For the top camera you may want light on for targeting, which is not really a vision operation. It may be that we need to integrate lighting control with the UI a bit more.

Other ideas to consider:
* A whole machine lighting system with actuators defined for certain functions. i.e. Default, Top Vision, Bottom Vision, etc. The actuators could be triggered when events for these functions are fired.

cri-s, 2016-10-30:
I have added actuator to camera as you maybe remember.
It was not intended to be error proof regarding configuration
and have some issues too.
It was done long time before vision pipeline was introduced.
On vision pipeline is not a problem inserting actuator on command that do image capture and disable it after capture.
Camera need string variable for actuator and global lightning actuator.

vonnieda, 2016-10-30:
I'm taking a little bit of a different approach. You can see the current work in https://github.com/openpnp/openpnp/tree/feature/camera-lighting if you'd like. Basically, adding a lightingColor property to Camera and adding support for it in ReferenceCamera via actuators. This will be supported by a new pipeline stage that can control the lighting along with options on the camera. Eventually this will be further integrated into the UI so you can turn lighting on and off as needed.

cri-s, 2016-10-30:
I don't see any advantage of setting colors and most cameras have problems
with cheap pwm rgb LEDs. Surely it looks cool.
Further it is a method to avoid having 3  actuators on each camera for
lightning.
But why not simply add string actuator to camera and pack lightning colors
into  second image capture stage that have setting time always on and
enable/disable lighting on uplooking camera and enable/disable global, not
first head lightning.

vonnieda, 2016-10-30:
@cri-s Having color on there is just an easy way to allow for additional
flexibility. It doesn't cause any harm for machines that don't support it.
This is in preparation for some more advanced loose part finding stuff I'm
planning down the road.

The way I have it set up now there are four possible actuators you can
define. One is simple on/off, so you can just use that one if you like. For
people who have RGB they can use the additional R, G and B actuators to
control.

The code you see now is not finished. I'm going to remove the bottom vision
part and just let that happen in pipelines. There will be a new stage that
is used to control lighting. I'll add a "global" option to it - that's a
good idea. Then you can have a stage that turns all lights off followed by
one that turns bottom vision on.

I will give some thought to adding a string based actuator. I considered
this earlier for the lighting color code, but I decided against it because
of the ambiguity it introduces. It may be a better way, though. I will
think about it.

Jason

cri-s, 2016-10-30:
Real PNP machine have 2 types of lights that are driven either by pwm
(0-10v) or setup using potentiometer. On every camera. This lights have 3
different and exclusive
On/off actors. Newer juki machines have 3 type of light for up looking cam.
Coplanarity check require 6-8 lights and require camera synchronized to
lights .
Now if camera is only used for fiducials and
Component detection,  upcamera only used for pins/resistor body, single
type of light is sufficient. It is then not possible to look at sign above
component for OCR, check pin1 or polarity. For loose feeder, using
permanent backlight and taking picture without and with downlight works for
pin1 detection and marking  if intensity of backlight is adjusted correctly
reducing a bit fps .
As probably you never switch to 2lights, and this mean automatically more
setup time required, why not simple on off actor per camera. Color look
cool and you have already the ui controls for that.
It looks workable, some errors. The biggest is thinking head cameras should
use head actuators for lighting. On head.moveSafeZ that is called basically
everywhere all head actuators are shut off. You surely don't want that
bahavior. Further you don't have thinked about having more then one head or
if there is additional led lightning just above the machine that could
disturb.

vonnieda, 2016-10-30:
Cri, Head actuators are not turned off during safe-z in my code. Perhaps
this is something you have added to yours?

Thanks for your comments, I'll take it into consideration as I develop this
feature.

cri-s, 2016-10-30:
Seen, you are right. It move only, sorry.
I supposed it switched off too because so slow.

dave-prosee, 2016-11-26:
@vonnieda 
My 2 cents:

Current architecture has dynamic info (e.g. x, y, z, rotation) available to the driver and controller specific commands defined on the driver in machine.xml. For camera's these dynamic values might be light intensitiy (per color), the angle a mirror has to have and maybe some more that I can not think of.

Capturing an image might involve a lot of tasks that can be wrapped up into something like "prepareImageCapture" and "postImageCapture". 

As openPnP is not aware of the hardware involved and the implications for what has to be switched it should abstract from these by just firing pre- and post-capture events, just like pick and place.
However the dynamic values should be supplied. Much like moveTo.

Some users like to keep it simple, others have more demands. All of these can, in my opinion, be solved analog to moveTo. 

A user that wants to set color might want to use variables like R, G and B. 
A user that just switches a light on an off could define a gcode with no variables, or a mix (using R for intensity).

Users that want to switch other lights off define pre- and post commands specific to their camera id's.

Some users with mirrors need to know the angle (e.g. with a mirror between two head on one camera) and can use an angle variable.
Others might just have one mirror on one camera and can use fixed codes per camera id.

Hence I am an advocate of:
1 - define a pre- and post command to use on the gcode driver
2 - defining variables which at first can be entered in the configuration screen per camera

Number 1 has the most value and is a prerequisite for number 2. 
Number 1 means adding two methods to GcodeDriver,  machine.xml definitions and some documentation. Number 2 is more work and I do not know all the implications yet.
So I proposed to start with 1.

vonnieda, 2016-11-26:
@dave-prosee Thanks for giving this some thought. So, if I follow, you are suggesting modifying the ReferenceDriver interface to include something like:

+ public void beforeCapture(Camera) throws Exception;
+ public void afterCapture(Camera) throws Exception;

Is that about right?

dave-prosee, 2016-11-26:
@vonnieda 
Yes. Exactly. I haven't read anything that can not be done in this way.

dave-prosee, 2016-11-26:
@vonnieda 
Jason, do you agree? Then I might have a go.

vonnieda, 2016-11-26:
@dave-prosee Working on a response now. I think we need to consider a different approach. Typing it up to see what you think.

vonnieda, 2016-11-26:
@dave-prosee Okay, I've been giving this some thought and I think I'd like to go in a different direction. I'm not committed to this yet, so feel free to discuss.

The main problem I see with the approach described is that while it works well for the GcodeDriver, it creates limitations for the other drivers. For instance, if someone is using the TinygDriver, they will need to modify the driver to add in the particular commands for their hardware.

I'd like to suggest an alternate approach that uses the Scripting and Actuator system:

I've been thinking a lot about using scripting to allow advanced functionality for events in the system. This would basically just be a set of specifically named scripts that get called, if they exist, when OpenPnP wants to do something.

I think this feature would be a good place to start using this. The idea would be that before we call Camera.capture we could call Scripting.on("Camera.BeforeCapture", camera) and pass the camera and then after the capture we call Scripting.on("Camera.AfterCapture", camera). Then it's just up to the user to define a script that does whatever they want. The scripts would be in a special subdirectory of the scripts folder called events, so if you call Scripting.on("Camera.BeforeCapture", camera) the scripting system would look for a file at .openpnp/scripts/events/Camera.BeforeCapture (with any valid scripting extension) and it would run that script. If none exists, nothing happens.

The scripts would primarily use the actuator system to do their work. So, for instance, a simple Camera.BeforeCapture.js might look like:

if (camera.name == 'TOP') {
  camera.head.getActuatorByName('LIGHTING').actuate(true);
  camera.head.getActuatorByName('MIRROR').actuate(true);
}
else if (camera.name == 'BOTTOM') {
  machine.getActuatorByName('LIGHTING').actuate(true);
}

And Camera.AfterCapture.js would just be the inverse, to turn things back off.

I know this sounds like it's maybe complex, but it would actually be really simple to implement, and it would provide a ton of flexibility. I'm thinking of slowly moving many of the events to this type of system so that we can avoid making the drivers more and more complex as people's use cases get more complex. I envision things like Feeder.BeforeFeed, Feeder.AfterFeed, Nozzle.BeforePick, Nozzle.AfterPick, Nozzle.Pick, etc.

What do you think?

vonnieda, 2016-11-26:
@dave-prosee I went ahead and added the internals to make this happen, but will wait on your feedback for making the actual camera changes. You can see the commit at https://github.com/openpnp/openpnp/commit/4f086bfd5e2bbe934b2af7699b11b830539bc887.

So now it's possible to run a pre-defined script just by calling Configuration.get().getScripting().on("Event").

dave-prosee, 2016-11-26:
@vonnieda 

I see, there are actually two issues then: light management and a preference for machine customization.
On a higher level you and I agree on the same solution i.e. marking the event and then like you wrote "it's just up to the user to define".... either in a script or in machine.xml.

I assumed, thinking you were advising Gcodedriver + xml a lot, that you wanted to go that way and finally deprecating other drivers. So my proposal was based on what I saw as the status quo. 

The examples from scripting do look more simple and intuïtive than the driver xml so I think you are right that this customization technique lowers the barrier to entry/use. And, assuming this is not an if question anymore but rather a when question, would that also imply that you want to get rid of the xml technique? Since I do not see situations that you can not cover with scripting this would eventually reduce code complexity too. I would strongly argue for that: but mainly from the users perspective that systems should simple and not a cathedral with too many options.

On what to do: in a commercial environment one would make a scenario analysis. Some customers are waiting and I feel lighting as something essential. A better solution takes longer and might make easier migration to new sw levels but in the meantime one does not offer a solution. And when you want to go all the way (meaning cleaning up all xml commands) what would cleaning up an extra two commands have for effect? Wouldn't migrating these new two be almost identical to e.g. moveTo()? So do we really ad extra work in the meantime to this desired transition?
Anyway, it probably best to make a bald decision as to not end up with two halfly implemented systems.
And you're the boss.

In the meantime I'll poke around the software to see if it brings any other ideas. t.b.c.

dave-prosee, 2016-11-26:
@vonnieda 

As you mentioned TinygDriver, I just browsed through it.
What is it, apart from checking the firmware version, that this driver can do better than the GcodeDriver?

dave-prosee, 2016-11-26:
@vonnieda 
I see comments crossed each other.
Very nice you went ahead. Discussion is always good but sometimes actions put these in a different perspective. I still would like to emphasize to choose between any one of the systems to reduce complexity overall since having two systems (scripting and xml commands) does not add functionality.

vonnieda, 2016-11-26:
@dave-prosee TL;DR: I'm going to go ahead and implement this and will have it done in the next 6 hours or so. 

Full explanation:

I used the TinygDriver as an example because it parses responses in JSON format and extracts properties from them. This is not something the GcodeDriver can do, or is intended to do. 

A little background is in order:

The driver system in general and the ReferenceDriver interface in particular are intended to make it possible to support any possible hardware configuration with OpenPnP. In the early days I wrote individual drivers for every controller, but it soon turned out that people wanted custom functionality (I want to use M810 instead of M8 for pick, for instance) so I wrote GcodeDriver.

GcodeDriver is intended to make it possible to support most of the Gcode like controllers out there through configuration rather than code. Most of the existing drivers can, and eventually, will be replaced by it. But there are many people out there who have written custom drivers for their own hardware where it would be too complex or impossible to run it with the GcodeDriver.

For instance, there are machines out there that use binary protocols, or require very precise timing, or talk over RS-485, ModBus, or TCP/IP. By implementing a ReferenceDriver for these machines they can use them with OpenPnP.

So, based on this, my guidelines for adding features to GcodeDriver are:
1. Is it generally useful to a large class of controllers?
2. Does it remain compatible with the rest of the drivers?

Another way of putting it is that GcodeDriver is intended to add flexibility to the system but it is not allowed to dictate how features work. The features must work within the ReferenceDriver interface, or the interface must be changed, and I don't like doing that very often because it breaks all the code in the field.

So, with that out of the way, my thinking behind the scripting stuff is to make it possible for users to add customization without systemic changes. I don't consider this to be a duplicate system. The driver is responsible for talking to the hardware at a low level, while the scripting interface is intended to allow the user to make customizations at a high level. The benefit of this is that it keeps the driver interface small and allows users to do extensive customization by using the existing interface. Actuators are the primary method.

By using Actuators and scripts rather than making driver changes, we can add features that are available to every machine / driver type, and not just GcodeDriver.

I hope that clears things up. If not, hang tight, I am just going to go ahead and add this in the next couple hours and we'll see how it goes. Camera lighting was the top item on my TODO list for this weekend, so now is as good a time as any to get it working. Adding the events is the simple part, the more complex part is fixing some sloppiness in the existing camera implementations that make it dangerous to do this work from the capture() method where it belongs. There's more info on that in the "Backlog" section at the top of this page, if you are interested.

dave-prosee, 2016-11-26:
@vonnieda 
Thanks Jason for your extensive explanation. I am missing some background info indeed.
And whilst testing scripting (I got it working) I found out there is one downside (as always according the law of never diminishing misery) in that scripting requires more work in advance. At least for Gcodedriver users. As the event is dispatched to scripting and only existing commands can be used in Gcodedriver you have to define actuators and gcode within the actuate command. So you have info in scripting, machine config and driver commands. 
I just say this for the sake of discussion and to help you consider all perspectives. But more importantly, I am pleased to hear lighting is soon to be finished ;-)

dave-prosee, 2016-11-26:
@vonnieda 
But what I saw as extra work has a pro too. All actuators show up as buttons in the actuators tab. Handy!

vonnieda, 2016-11-26:
@dave-prosee This feature is now in. More information at https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2016-11-26. Please give it a try and let me know how it goes for you.

closed 2016-11-26 by vonnieda

dave-prosee, 2016-11-27:
Hi Jason,
All works fine. Thanks for the change. 
As insight progress, also with testing now, I have some findings.
The capture event is relatively late. E.g. when you're doing an auto setup on a strip feeder the camera is already positioned. And when turning lights on my camera has to adjust. I can handle both by leaving the top light on as much as possible and implementing some delay when bottom vision is needed. The system does give me a lot of flexibility now and almost on the fly changeable (don't need restart for javascript change). Since I leave the toplights on to circumvent lightlevel-adjusting-delays (top camera is used more) I now manually switch it on before starting to work. The setup event is too early (controller not connected) to use this. So yes, a good and very nice improvement. 
Further improvements could be to have events like "connected", "going for part alignment", "going to a feeder".  Thanks!

vonnieda, 2016-11-27:
@dave-prosee Glad to hear it working. I noticed the same thing with my testing last night. My camera sends really bad frames for about 750ms after turning on the lights. I think this can be alleviated by using manual exposure control, but that is going to require completion of another big project.

As for turning the lights on manually, you could just add this to your GcodeDriver ENABLE_COMMAND.

dave-prosee, 2016-11-27:
@vonnieda 
Off course, ENABLE_COMMAND. Thanks!
I am at the point of completing my machine. Probably two weeks or so. A lot of things fall into place. Some require a refit (vacuum is making a hell of a noise, lights always on are getting too hot, nozzle to hollow steper axis needs precise connection,..). But the more I work with openPnP the more happier I become.

commit 6f30637b (2016-11-26): Fixes #355 by adding scripting events before and after a camera capture happens. The scripting events can be used to turn lighting on, or anything else.