openpnp/openpnp #357: Feeder System Improvements

vonnieda, 2016-11-01:
# Draft, Please Withhold Comments Until Complete

There are several new ideas for feeders floating around, along with a couple pull requests, that require modification of the feeder system to support. This issue is intended to capture the requirements for the changes and provide a point for discussion about them before the changes are made.

The primary driver is @tracklogic's PR (#344)  for a slot auto feeder. This is a new type of feeder that can be moved around on the machine without having to be completely reconfigured. This is common in commercial machines, as well. The concept is that you have a number of feeders that can be readily attached to the machine and instead of removing tape and reels from feeders, you remove the entire feeder at once and swap in a new one.

There is additional discussion about this feature in https://groups.google.com/d/msgid/openpnp/aa380a0e085732e156dee535fa5773b6%40mail.mattbrocklehurst.co.uk but I will attempt to capture the ideas here.

A secondary driver is the re-addition of the canHandle(Part) API and possibly the re-addition of the feed(Part) API on Feeder. These are to support Frantz's improvements to the loose part feeder which allow multiple parts to be used in one bin.

Another thing to consider is the concept of feeder banks, where an entire section of feeders may be removed and replaced with a different set.

In addition, there is a known bug in the FSM based JobProcessor code that causes feeder errors to be swallowed. 

In summary, the goals are:

1. Movable feeders.
2. Feeders that can feed multiple distinct parts.
3. Feeder banks
4. JobProcessor issue that swallows Exceptions.

Each of these topics is discussed below, and finally at the end we have a set of proposed changes to the code to support this work.

# Movable Feeders

Requirements:
1. It should be possible to have a set of fixed, defined slots that can hold a feeder.
2. Each slot has a pick location that, in general, never changes.
3. It should be possible to notify the user that a particular configured feeder is needed and ideally be able to recommend which slot to put it in.

# Multiple Part Feeders

# Feeder Banks

# JobProcessor Issue

cri-s, 2016-11-02:
Please leave the option to mount and unmount subfeeders to the wizard page and not makit it a global feature needed for all feeders just because one needs it.
Continuing placing without refill works to a certain degree.
How do you place resistors near Ethernet connector when ETH is already placed? Pls retrain concept one feeder, one component.

Please r

tracklogic, 2016-11-02:
I keep thinking perhaps we need a feeder manager singleton, it can be passed the complete job the job processor intends to run and after each step of the job being processed (and maybe on any feeder errors) the job processor notifys the feeder manager. 

Before the job runs, the job processor passes the job over, and the feeder manager can work out the best place for all the feeders to be and also the best order for the job i.e. if we need 40 different parts, and we've got 40 feeders but only room for 20 feeders (20 feeder slots defined), we should reorder the job for the minimal amount of feeder changes.

After each step of the job the job processor will notify the feeder manager and this can then check if we need any feeder swaps, and if so, inform the user before allowing the job processor to continue?

Might be a crazy idea though :)

vonnieda, 2016-11-02:
@cri-s 

* Yes, my goal is for this to all be encapsulated in a Feeder implementation, not to make major changes across the board to support it. We may have to make minor additions, and that is what we are here to discuss.

* Placing short components near tall: This should be handled by the planner. I addressed this in comments in the doPlan function. Based on feedback from users, most machines will have enough Z height to deal with this without special requirements, but one option is to add height sorting into the planner.

* We will probably stick with one feeder one component, but I am open to changing it if there is enough value. Can you explain your concerns with it so we have that data to consider?

@tracklogic I agree, to a point. I think that at least we need a way to tell the feeder more about the job. I mentioned above "I think we're going to need to have the processor pass some kind of context to the feeder that says how many of each part are still needed for the job. This will allow the slots to make intelligent decisions about which feeders to load. Alternately, this may be something we want to handle at a higher level."  I want to spend some more time thinking about how useful this will be in general, or if it's really just to support your use case.

Based on @cri-s's recent comments in the FSM thread and some musing I've been doing, I'm starting to think there would be value in having a job context that contains a good representation of the state of the job.  It would contain things like the list of placements, each placement's status, etc. This would be mostly read only and could be passed around to various interested parties, such as feeders, part aligners, etc. In particular, we really need this data to show status in the job window.

I think we probably need this whether we make a feeder manager or not. It would just be matter of whether it gets passed to the feeder manager or the feeders themselves.

To @cri-s and @tracklogic: Before we get too far off into the weeds, I'd like to ask that we stick to requirements for now rather than implementations. Let's capture requirements for the things we want to do and we'll sort out how we're going to do it later.

cri-s, 2016-11-02:
If one feeder support multiple parts, feeder enabled need to per part too. Next may by you want having spliced reel feeder, some machine can that. 5 part this component, 10 part that component, ...
For reset feed counter, this need per part in future, not feeder, because I'm sure if you introduce feeder that can do multiple parts, tray feeder with different part refilled at different time are one of first requests. 
It is possible. Feeders need give out parts that it Support and one must iterate that feeder with returned parts,one by one.
Big change for no real gain only complications. This my opinion.
. 
For feeder manager, you don't need it as I do it on long time and Jason have seen it too and code was written from Ami.
It changed placement.type to differentiate about already placed, todo, done and just need some rewind and forward icons or scripts.

vonnieda, 2017-01-03:
Broke this out into other, more specific, tickets, so closing this one.

closed 2017-01-03 by vonnieda