openpnp/openpnp #358: Create Github Tutorials

vonnieda, 2016-11-03:
In an effort to make it easier for people to get involved with OpenPnP, we should create two tutorials to help people understand Github.

1. A step by step tutorial (video or Wiki with screenshots) showing how to fork OpenPnP, make a change (preferably referencing an existing issue), commit it, push it up to the fork and issue a pull request.
2. A step by step tutorial (video or Wiki with screenshots) showing how to create a new Github repo and upload a simple project to it. This will help get people started with Github when they have interesting things to share, such as feeder and machine designs.

Further discussion at https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/rJ1DZniw8eY/zrp-MmOYAgAJ

dave-prosee, 2016-11-03:
Hello Jason,

I wonder what the real issue is here and what should be done.
Forking and creating Github repo's are already covered by Github itself. See e.g. the homepage of Github (github.com); you'll find two big buttons "Read the guide" and "Start a project".

Maybe my own experience, and of others, can help in finding what the real issue is. You are very active and I think for this issue you should first find out why people behave like they do. Being so active has a price too, as others take things for granted or feel the are maybe not so competent. 

My own experience with some other open source (ESP8266/Arduino) made me clear that everything happened in Github there. Therefore people easily made new issues on Github and were already "closer" to the software. It took me some effort but I got my request only done by making a pull request. The owner was active too but made me come more towards Github. In this particular case I was already having some knowledge of the Arduino IDE (as well as Eclipse) so that made it more easier.
As I also have Java experience it would not be so difficult to participate in OpenPnP but I still have some reluctance. The reluctance stems from the fact that I don't have an overview of the software architecture and the way I could change little things without the change becoming too difficult.
Also, as I already mentioned, the number of communication channels for OpenPnP is big. I think this lowers barriers to participate but with a cost that people can easily stay in their comfort zone.

Anyway, this was my experience. I hope some more people will tell how they experience things so you have a better insight in the reasons. For me software architecture is more of an issue to step into participating.

Best regards,
Dave

vonnieda, 2016-11-03:
Hi Dave,

The purpose of these two tasks are to make it easier for people who want to participate but don't understand Github. There are two primary issues I want to solve:

1. Someone knows how to code but has never worked with Git, Github or maybe source control at all. This often results in people sending me code as attachments, zip files, or not sending it at all because they are confused.
2. Someone has created something useful or interesting, such as a machine or feeder design, but has never worked with source control and doesn't know how to get the files on the web for people to see. Again, this often ends up with files being passed around via email or temporary links.

These are cases I've seen over and over since starting this project. 

My goal with these two tasks is to create two very simple tutorials to help people get their feet wet with Github. I realize that Github has their own documentation for this stuff, but I feel it assumes a level of knowledge about source control that a lot of people don't have. And in addition, I think that by having the tutorials be specific to OpenPnP it will be easier for people to understand how it all fits together.

I'm open to suggestions, of course. If you think there is a tutorial out there already that solves these problems, please provide a link and I'll have a look.

dave-prosee, 2016-11-03:
Hi Jason,

Sorry if my style is too direct. I am Dutch and my culture is known to be fairly direct and to the point. 

It must be stated that I am new to the OpenPnP scene and don't have the experience of the cases you mentioned. On the contrary I have seen many (business) initiatives not leading to the desired effect mainly by not researching the real reasons.
That, together with my experience so far with OpenPnP and other open source it drove me to write a reaction (I think there is more. And for me it isn't Github knowledge).

I do agree that it lowers the bar for people if you have  specific manuals. On the other hand, people with skills to program can, most of the time, handle generic issues and more abstract tuition.  
As I see myself in the latter category I find myself more in need of a guide into the already existing software. Or in a more tangible way: what if I were to have an idea for e.g. nr of solder paste puffs according area size. Where would I start? Or the many requests on editing a job,...how/where could you start (besides source control)?
Another example: I have read posts were you claim something should be a singleton, that feeders should be update more frequently, and so forth. For me this is too far ahead and I wish I could have some documentation/guide to get up to par. Then I would feel I could add my value by participating.

Please read my good intentions here as I've given my experience and view. Equally I am interested in other experiences to balance my view. 

Cheers,
Dave

vonnieda, 2016-11-03:
Hi Dave,

You aren't too direct. It's all fine :)

Have you read through the Developers Guide? It includes basic information
about the software architecture and how to find the various major
components of the system.

https://github.com/openpnp/openpnp/wiki/Developers-Guide#system-architecture

In my opinion, the best way to learn about a system's architecture is to
get in and try to change something. I believe there is a certain amount of
hands on work that must be done before you can really understand a large
piece of software and I don't think that a large guide is a good substitute
for that experience. Whether good or bad, OpenPnP is a relatively large and
complex codebase and it takes some fishing around before you can understand
how it all works.

With that being said, I have been told by many other people who have worked
in the code that it is well organized and well defined. If you have
experience with OOP systems I think you'll find the code is pretty easy to
follow.

If there is one "trick" I will mention it's that you should consider the
machine.xml file to be the first place to start for any work. If you want
to know how a Nozzle works, look at the machine.xml, find the nozzle and
find the class it uses. You'll find a reference to
org.openpnp.machine.reference.ReferenceNozzle and this is the concrete
class that does the work of a nozzle. The same goes for the driver, paste
dispenser, bottom vision, cameras, etc. Everything starts in the
machine.xml.

If there are specific issues you are interested in working on and would
like some guidance, please let me know. I'd be happy to point out where in
the code a person would start to look for a given issue.

Jason

dave-prosee, 2016-11-03:
Thanks Jason,

Yes, I have read the developers guide but maybe not too good ;-).
I will have a look later tonight or tomorrow and let you know.

robermeyer, 2016-11-03:
Jason,
You are proposing the right thing.
My experience with this is
1) The user thinks a fork or contribution is over their head.  They just
need a little push and they will do great things.
2) They have no clue and will never have.  They should stay away from
github!
3) They want somebody else to do all the work.
4) They know it all and get out of their way.

You have tried to support all the groups even when its obvious they don't
deserve it, in some cases.
I hope we can help group 1 the most.  Seeing someone realize "they can do
it" is most rewarding.

vonnieda, 2016-11-04:
First tutorial is done: https://github.com/openpnp/openpnp/wiki/Your-First-Pull-Request

dave-prosee, 2016-11-04:
Hi Jason, Nice tutorial. I executed it and troubled you now with a pull request. ;-)
Very meticulously done. 

Please have a look after the headline "Push to Github". I think you occidentally pasted the same screen shot ( screen shot 2016-11-04 at 12 58 14 am)  there for the second time instead of the push.

Best regards,
Dave

dave-prosee, 2016-11-05:
I promised to come back on the developers guide, so here I am. Last night I went through the developers guide again and your are right. The software architecture is more visible to me now. I wanted to start changing things for a better understanding and needed an IDE. I had Eclipse as well as IDEA on my machine. As I didn't wanted to be bothered by the Maven stuff I chose IDEA. Then made some mistakes I guess and ended up with some problems making/compiling that I am now trying to correct by a fresh clone and new install of IDEA (bit harsh action, but I want to get away from installation details and follow the developers guide). Anyway, I should say that there is indeed no need for additional explanation of the architecture. This guy however has something to learn with his IDE. 
Greetings,
Dave

dave-prosee, 2016-11-05:
And some more: I switched to Eclipse now with better results that I would like to share.

Eclipse is now on version Neon and has Maven integrated. That makes a lot of the Eclipse tutorial outdated. Since I don't feel that experienced to lead others I just give my remarks here (instead of changing the tutorial):

With reference to the Step by Step hints:

Current: Go to http://www.eclipse.org/downloads/ and download "Eclipse IDE for Java Developers" for your platform. When the download completes, unzip or install Eclipse and start it up.
>> I hesitated between "for Java" and "for Java EE". This might be a nobrainer for some but for whom are using the tutorial maybe not. I chose "for Java". Also I did not have to make a platform choice. Probably the website picks my info.

Current: Eclipse will ask you where to store it's workspace. The default is fine. You can click "Use this as the default and do not ask again." if you don't want to be bothered every time you start Eclipse.
>> No comment

Current: When Eclipse starts you will see a Welcome screen. Close the Welcome screen by clicking the small X in the tab.
>> The welcome screen just about covered all of Eclipse and looked like a menu instead of a Welcome screen. I just clicked the X in good trust.
 
Current:    You should now see the Package Explorer tab. Right click in this area and select Import.
                 On the Import screen, expand "Git", select "Projects from Git" and click Next.
                 On the Select Repository Source screen select "Clone URI" and click Next.
                 On the Source Git Repository screen enter https://github.com/openpnp/openpnp.git in the URI field and click Next.
                 On the Branch Selection screen select the "develop" branch and unselect all other branches, then click Next.
                 On the Local Destination screen enter the directory where you would like to store the OpenPnP source code. You can keep the default if it's okay. Then click Next.
                 Eclipse will now download the OpenPnP source code. This may take a few minutes.
                 When it finishes, on the same screen select the "Import as general project" option and click Next.
                 On the Import Projects screen accept the default project name of "openpnp" and click Finish.
                 The openpnp project will now appear in the Package Explorer. Right click on the project "openpnp" and in the menu select Configure -> Convert to Maven Project. Eclipse will download any resources that it needs and build the project.
>> As Eclipse now includes Maven and trying to be compatible with the "Your first Pull Request" tutorial I just imported the Maven project by refering to my GitHub clone.    

Current: Expand the openpnp project and look for the src/main/java folder. Under this folder expand the org.openpnp folder and look for Main.java.
              Right click Main.java and from the menu select Run As -> Java Application.
              After a few seconds OpenPnP will start. Close OpenPnP.
              You can now edit files in Eclipse and the changes will automatically compile. To run OpenPnP again with your changes just click the green Run button in the toolbar. The button is a green circle with a white right facing arrow in it.
>> This worked right away. Even debugging worked right away (which I was having trouble with using IntelliJ IDEA and the main reason to change to Eclipse).

In the meantime I found myself a challenge for which I will open an issue since I'm in need of feedback on the solution.

Cheers.

myGitV, 2016-11-05:
Request to add one more point to 'Your First Pull request' Tutorial.
After a small period, user's github fork needs 'update' from main openpnp commits.
How to periodically keep user's local fork updated in sync?
e.g. message like
"This branch is 3 commits behind openpnp:develop"
needs attention.

Solution to this can be found on GitHubs tutorial / documents
But if added to 'Your First Pull request' Tutorial, keeps things neat and a beginner will not digress away initially. (Sounds like spoon feeding, but nice to have at start)

vonnieda, 2016-11-05:
@myGitV I agree. I have "pull merged changes" and "create a feature branch"
on my TODO list for the tutorial for this weekend. Coming soon! :)

Jason

pfried, 2017-01-05:
Hey Jason,

altough i do like your efforts to lower the bar for new people to come in i feel that it is not the best idea (i scrolled down the issue, so i learned later that you already created the guide) to have this guide. Why? Because there is tons of material on the issue of creating a commit / pull request / contribution.

People really wanting to contribute will need to get the basics of using git. Your guide could help here and this is your intention but i feel that you are trying to teach them how to use Github instead of Git and are using it only as a means to an end. This is most likely a bad idea. There are so many cornercases your tutorial cannot cover like git configuration , errors during committing and so on. I feel that really learning git first would be the right way. Creating pull requests is sugar on top of that.

This project is about OpenPnP and not about teaching people how to use Git/Github. As a project lead you should really try to concentrate on the core "business" of the project.

Creating the guide is not the only work that needs to be done. Things tend to change and you will need to keep track of it. Also it does not cover everything (altough it is very detailled) so people will either ask or get some tutorial / guide somewhere else anyways. I would rather include a short guidance into the contributing.md and point to the very good and also very accesible tutorials of Github / Atlassian / ...

These are just some thoughts of me...

Best, Friedrich

vonnieda, 2017-01-05:
Hi @pfried,

Thank you for the feedback. I appreciate it, although I will have to respectfully disagree.

I explained my reasoning in https://github.com/openpnp/openpnp/issues/358#issuecomment-258152618. Since posting this tutorial I have had several very good pull requests where people referenced the tutorial and said it helped them get their first pull request done. That's a win in my book!

The CONTRIBUTING.md does reference our Developers Guide, and that has links to Github's documentation along with our own.

I have been running this project for over 5 years, and I invest thousands of hours per year into it. I have a unique perspective into the pain points that people hit when trying to contribute and adding this tutorial seems to have helped alleviate some of that. I feel that is time well invested.

robermeyer, 2017-01-05:
> I explained my reasoning in #358 (comment)

Jason,
Just last week I spent a few hours going over your tutorial and doing some
Github work. I think your tutorial was a great help.  Yes I could search
all over the web and find yet another 50 tutorials unrelated to OpenPNP.
It was very helpful to have somebody who knows what I am trying to do going
through the same project I am and explaining things as I go.
It even helped me on a totally unrelated project I had to work on.  They
had no tutorial at all.
I give you the usual thumbs UP.

vonnieda, 2017-01-05:
Thanks @robermeyer, glad it helped, and I appreciate you saying so!

issus, 2017-01-05:
Anything that gets more people contributing is a win in my book :)

closed 2017-04-13 by vonnieda