openpnp/openpnp #359: actuators

cri-s, 2016-11-03:
Improve actuator spi :

() return integer value. Example ADC value.
If actuator don't support it, return -1.
(Double) used as dwell delay, for dispenser, glue
(-double) emit m code with abs value, optional p
Example -800.33. Emits m800p33 -800 emits m800
(Int) used to pass value, pwm for light, ...
Possible implementation is S gcode. -int could T gcode
(Boolean) as is

cri-s, 2016-11-03:
i mean that  public int actuate(void) throws Exception;  should be valid for all acutators, and 
ReferenceActuator.java should provide default operation implementation of returning -1 and provide
public variable value that can be used to set returned value as example.
this then allows:
vac = nozzle.getHead().getActuator("vac"+nozzle.getName());
if(vac() > limit) do_something();

The same should work on pwm, as example if setting light actor to 80% setting
light(80)  light() should return 80; same for boolean, if setting actuator on, it should return 1, otherwise 0. If no boolean or level was never used, -1 should be returned based on the obove definitions.

vonnieda, 2016-11-04:
Okay, I like this idea. How about this:

```
Integer actuate(void);
Boolean actuate(boolean value);
Integer actuate(int value);
Double actuate(double);
```

By using Integer, Boolean, Double instead of int, boolean, double we can return null when the value is not known. So, instead of returning 0, 1, -1 for boolean we can return null, true or false. This makes the API less confusing.

Will this work for you?

cri-s, 2016-11-04:
Only actuate(void) should return int, the other need to return void.
If you want find out the value for light as example, you cannot send down another value just to know the result and the restore the old value. On boolean it could work, you could send down null.
If that api change is too confusing for you, its better I close the ticket.

closed 2016-11-04 by cri-s