openpnp/openpnp #369: Glue dispenser is throwing an exception when no dispenser defined, even if not using glue.

vonnieda, 2016-11-12:
The way the glue processor works caused it to run for every job, even if it's not being used. In the glue processor it attempts to get the default dispenser and if there isn't one it throws an exception, stopping the job, even if the user is not trying to do any glue.

In addition, the JobPanel code is not correctly detecting when there is no glue processor defined, and it throws an NPE. It should check if there is not glue processor defined and ignore the glue in that case.

Temporary workaround is to add a dispenser to the machine.xml:

```
    <head...>
            <paste-dispensers>
               <paste-dispenser class="org.openpnp.machine.reference.ReferencePasteDispenser" id="D1" name="D1">
                  <head-offsets units="Millimeters" x="0.0" y="0.0" z="0.0" rotation="0.0"/>
               </paste-dispenser>
            </paste-dispensers>
    </head>
```

With the dispenser defined the code will work, and nothing will be dispensed if the Glue option is not set on any placement.

tracklogic, 2016-11-14:
Hi Jason

I had a quick look at this ... 
src/main/java/org/openpnp/gui/JobPanel.java

```
            if(jobProcessor == null || jobProcessor == Configuration.get().getMachine().getPnpJobProcessor())
            {
               // Run the glue dispense processor first, this will deposit glue ready for any component placements
                jobProcessor = Configuration.get().getMachine().getGlueDispenseJobProcessor();
	}
```

Should be something like...

```
            if((Configuration.get().getMachine().getGlueDispenseJobProcessor()!=null) && (jobProcessor == null || jobProcessor == Configuration.get().getMachine().getPnpJobProcessor()))
            {
               // Run the glue dispense processor first, this will deposit glue ready for any component placements
                jobProcessor = Configuration.get().getMachine().getGlueDispenseJobProcessor();
	}
```

Do you want me to have a look at it... guess i should as i was the one that broke it :(

vonnieda, 2016-11-14:
@tracklogic Yes, please. That would be very helpful!

vonnieda, 2016-11-16:
Fixed in #370

closed 2016-11-16 by vonnieda