openpnp/openpnp #376: independant Z axis impacts Gcodedriver (when XY move to subdriver)

dave-prosee, 2016-11-25:
A 2nd Z axis gives double tool offset in action. The reason why and first actions are mentioned below.

Having a second independent Z axis (5th axis) often means having a second controller i.e. interface because not all controller boards support 5 axis.
Homing then becomes something you have to manage with the main GcodeDriver and a subdriver.
Because you want to home all Z axis before any X or Y move this means that the second Z axis is driven by the main driver and the other axis by the a subdriver. This is because the main driver actions are executed first before any subdriver action.

The Gcode driver is one driver used as main and subdriver. It first estimates the desired location for the given headmovable by correction for the headmovable offset. Then it decides whether actions are needed based on machine.xml, given location and current location. If needed it executes the gcode actions and calls itself for all subdrivers it has.
  
Currently, if no processing is needed the Gcode driver return without giving control to subdrivers.
This happens at two points: when no axis is defined for current headmovable or when no move is necessary for current headmovable.
Tweeking machine.xml could force the program beyond these points but reveals a second problem that during this proces the desired location to move to has already been corrected for the offset of the headmovable. So when reaching the next execution of the Gcode driver (e.g. in a subdriver) the desired location is again corrected for the headmovable offset. Any switch between headmovables (e.g. camera and nozzle) managed by different (main or sub) drivers will thus result in a physical move that is a multiple of the defined offset. With more (sub)drivers this error increases.

The fix this seemed to be twofold at first:
- allways give control to subdrivers, even when no action is necessary in the current/main driver 
- don't build offset on offset, use the original desired location or build on top of executed increments per axis so that machine.xml really controls, see additional steps below. Another solution is using a main subroutine, holding location and calling main and subdrivers (for now I do the first)
Not that theoretically subdrivers could call other subdrivers. The code calls itself and is not aware of how deep subdrivers are subbed.

This fix makes it possible to only have a second Z axis definition on the main driver. But removing X and Y (and Z1 and C) reveals something new: the coordinates are not updated in the user interface neither under the hood. Any new action is based on the home location (0,0,0,0). This means an increment of 1mm on e.g. X makes the machine jump to 1,0,0,0. 
Restoring X and Y (and Z1 and C)back creates a workable solution but is not ideal.
@vonnieda can you give me a hint as to how results are stored and retrieved for displaying. Further more I propose to cover this issue as a second step, i.e. second pull request so we have some progress.

Research made think of the following additional steps:
- when any action is given (i.e. in command) we should store the new location for that axis. This should be done before the next point:
- a regex containing a feedback location should be able to overwrite the calculated location. So every gcode given in any main/subdriver should be able to calculate first and optionally consume the feedback location.
@vonnieda : your view please

First discussions can be found here: https://groups.google.com/forum/#!topic/openpnp/MfB-RH5e1wo

dave-prosee, 2016-11-25:
@vonnieda 
Hello Jason, 
I made a PR for first part.
When syncing with the latest code I got some merge remote-tracking commits that I could not get rid off.
See the first two commits (befcd10 and  	d4ad9b1). How do handle these?

Cheers,
Dave

vonnieda, 2016-11-26:
#377 has been merged, fixing the first issue. I've commented on that PR that we can likely fix the second issue by modifying getAxis, which @dave-prosee is going to look at.

vonnieda, 2016-11-26:
Fixed by @dave-prosee in #377 and #378

closed 2016-11-26 by vonnieda