openpnp/openpnp #379: Add an option requiring homing before machine usage.

vonnieda, 2016-11-27:
In most cases it's desirable to home the machine before moving it to avoid crashes. There should be an option that causes the machine to throw an exception if it is moved before homing.

The ideal place to do this would be in the head, but currently moves go directly from HeadMountable to driver. We could either put a home check in every HeadMountable or change the HeadMountable contract to push moves through the head. The latter option would improve the situation for head offsets and soft limits, but it's a pretty big change.

vonnieda, 2017-04-13:
I'm closing old issues that have seen little interest. PRs welcome.

ldpgh, 2017-04-15:
1. Change color of the enable/connect-button to YELLOW when turned on.
2. Change color to RED after "homing"

Though separate PR has initiated the files were added to PR #520 
It's just an proposal ... drop it if you like.

ldpgh, 2017-04-15:
Files modified
- openpnp/src/main/java/org/openpnp/gui/MachineControlsPanel.java
- openpnp/src/main/java/org/openpnp/gui/support/Icons.java
New file
- openpnp/src/main/resources/icons/power_button_off_nohome.svg

ldpgh, 2017-04-15:
Btw. for me a real topic, because my machine-controller runs "home" right at "enable". Forgot to often to press the home button and got few crashes.

vonnieda, 2017-04-15:
I think it's a great way of doing this. Nice work :)

vonnieda, 2017-04-15:
Fixed in #520

closed 2017-04-15 by vonnieda