openpnp/openpnp #383: Gcode vacuum pump start/stop at beginning/end of a job

bzhex, 2016-12-21:
It would be nice to have in Gcode driver some commands like "JOB_START" and "JOB_END" where user can send Gcode command for start/stop vacuum pump when job is running. I dont see why a vacuum pump should start on "ENABLE_COMMAND" for axample. If I have to do test movements, calibration etc, i do not need vacuum pump started. Also in Jog panel would be nice to have 2 buttons - START / STOP VACUUM.

vonnieda, 2016-12-21:
@bzhex There are PUMP_ON and PUMP_OFF commands that are sent when the pump is needed. Will these suit your use case? https://github.com/openpnp/openpnp/wiki/GcodeDriver:-Command-Reference#pump_on_command. Using these commands guarantees the pump is only turned on when it's needed.

For testing the pump, it's best to just use the test pick and discard buttons. It's difficult to have a specific vacuum on and vacuum off command since these are very different depending on machine configuration.

bzhex, 2016-12-23:
Missed that commands. Worked great.

closed 2016-12-23 by vonnieda