openpnp/openpnp #384: Null pointer error

dzach, 2016-12-21:
Hi
I have a prebuilt binary already configured and running without problems. Now I'm trying to test the new features by running the code in the **_develop_** branch: The code compiles without errors but I'm getting the following when running the program:

```
./openpnp.sh     
java.lang.NullPointerException
        at java.awt.EventQueue.getCurrentEventImpl(EventQueue.java:848)
        at java.awt.EventQueue.getCurrentEvent(EventQueue.java:842)
        at javax.swing.JComboBox.fireActionEvent(JComboBox.java:1243)
        at javax.swing.JComboBox.contentsChanged(JComboBox.java:1332)
        at javax.swing.AbstractListModel.fireContentsChanged(AbstractListModel.java:118)
        at javax.swing.DefaultComboBoxModel.setSelectedItem(DefaultComboBoxModel.java:93)
        at javax.swing.DefaultComboBoxModel.addElement(DefaultComboBoxModel.java:131)
        at javax.swing.JComboBox.addItem(JComboBox.java:716)
        at org.openpnp.gui.components.CameraPanel.createUi(CameraPanel.java:138)
        at org.openpnp.gui.components.CameraPanel.<init>(CameraPanel.java:61)
        at org.openpnp.gui.MainFrame.<init>(MainFrame.java:337)
        at org.openpnp.Main$1.run(Main.java:86)
        at java.awt.event.InvocationEvent.dispatch(InvocationEvent.java:311)
        at java.awt.EventQueue.dispatchEventImpl(EventQueue.java:756)
        at java.awt.EventQueue.access$500(EventQueue.java:97)
        at java.awt.EventQueue$3.run(EventQueue.java:709)
        at java.awt.EventQueue$3.run(EventQueue.java:703)
        at java.security.AccessController.doPrivileged(Native Method)
        at java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)
        at java.awt.EventQueue.dispatchEvent(EventQueue.java:726)
        at java.awt.EventDispatchThread.pumpOneEventForFilters(EventDispatchThread.java:201)
        at java.awt.EventDispatchThread.pumpEventsForFilter(EventDispatchThread.java:116)
        at java.awt.EventDispatchThread.pumpEventsForHierarchy(EventDispatchThread.java:105)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:101)
        at java.awt.EventDispatchThread.pumpEvents(EventDispatchThread.java:93)
        at java.awt.EventDispatchThread.run(EventDispatchThread.java:82)
```
Being ignorant about java and seing the line:
`java.security.ProtectionDomain$JavaSecurityAccessImpl.doIntersectionPrivilege(ProtectionDomain.java:76)` ,
I tried to run it with 'sudo' but got the same results.

Any idea about what the problem might be?

my system:

```
Kubuntu 16.04

$ uname -a
Linux desk 4.4.0-36-generic #55-Ubuntu SMP Thu Aug 11 18:01:55 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux
4xAMD A6-3650 APU with Radeon(tm) HD Graphics

$ java -version
openjdk version "1.8.0_111"
OpenJDK Runtime Environment (build 1.8.0_111-8u111-b14-2ubuntu0.16.04.2-b14)
OpenJDK 64-Bit Server VM (build 25.111-b14, mixed mode)

```

Thanks!

vonnieda, 2016-12-21:
@dzach Please update your code. The error was fixed this morning. See https://groups.google.com/forum/?utm_medium=email&utm_source=footer#!msg/openpnp/Z8NfqGBWUjc/x5ayZgaQEQAJ for more information. Please reopen this ticket if it is not resolved.

closed 2016-12-21 by vonnieda

dzach, 2016-12-21:
The code runs fine now.
Thanks!