openpnp/openpnp #386: Multi Window Mode Improvements

vonnieda, 2016-12-22:
Based on PR #385 by @ldpgh which adds multi window mode, there are some improvements that should be made to make the feature complete:

* Each window should remember it's size and position on screen. Look at the componentListener on line 716 to see how this is done.
* Currently on Mac the menu is not visible unless the main window is selected, and it should be visible when any window is selected. I think using dialogs instead will fix this but I have not tested it.
* If a user closes one of the windows there is no way to get it back without restarting the program. I suggest modifying the Window menu to have three options: Cameras, Main and Machine Controls and make clicking on any of these options bring the window up.

ldpgh, 2016-12-22:
"Currently on Mac the menu" ... does it refers to the new "Window"-menu?

ACK for  1 and 3 .

vonnieda, 2016-12-22:
@ldpgh The entire menu. On Mac the menu is global. It is attached to the top of the screen, not to the top of an individual window. Whatever window has the focus is what menu is shown at the top of the screen. Since all three windows in the new system are implemented as top level windows but only one has a menu, the menu is only shown for that one.

Here are some screenshots to explain:

Main window focused:

[image]

Cameras window focused:

[image]

Machine Controls window focused:

[image]

If you don't have access to Mac, then don't worry about that one. I'll take care of it eventually.

ldpgh, 2016-12-22:
Got it.

ldpgh, 2016-12-22:
Before I think of 1 and 3 ...
... could you do a quick'n dirty test on Mac with

```
JDialog frameCamera = new JDialog(); //Frame("OpenPnp - Camera");
...
JDialog frameMachineControls = new JDialog(); //Frame("OpenPnp - Machine Controls");

```

For me the behavior is the same, if the window title is set afterwards.

vonnieda, 2016-12-22:
Looks like this does the trick:

JDialog frameCamera = new JDialog(this, "OpenPnp - Camera");
JDialog frameMachineControls = new JDialog(this, "OpenPnp - Machine Controls");

I found the info here: http://stackoverflow.com/questions/16175884/what-is-the-best-way-to-make-an-os-x-screen-jmenubar-work-consistently-across-wi

Haven't done extensive testing yet, but at first glance it seems to work okay.

vonnieda, 2016-12-22:
Okay, that didn't work because they were modal, but this does:

```
JDialog frameCamera = new JDialog(this, "OpenPnp - Camera", false);
JDialog frameMachineControls = new JDialog(this, "OpenPnp - Machine Controls", false);
```

Checking the change in now.

vonnieda, 2016-12-22:
Change made here: https://github.com/openpnp/openpnp/commit/9856567e5e0d1fcb37726dd89a2c3181c45995ca

ldpgh, 2017-01-29:
My fork synchronized to openpnp/openpnp/develop @ 29.01.2017

---

(notes for me) current status in the repo
- size of the sub-widgets is already kept ... done
- position is lost
- local program changes to handle widgets individual seems to be lost
           - need to scan my harddisc, maybe 
- restart of application required if window mode has been changed
            - should not be necessary
            - check local changes
- current implementation does not allow to bring main window to the front
     - check JDialog vs. JFrame and modeless vs. modal mode

ldpgh, 2017-02-05:
More notes ....
- maybe the links associated with http://stackoverflow.com/questions/10366404/putting-jmenu-on-the-mac-menu-bar let return JFrame for OS/X ... need to be checked.
- maybe a console/log window could be added as well (eg. when debugging multiple webcams allocated by other program)

vonnieda, 2017-02-05:
@ldpgh The menu issue has already been resolved. Notice the checkboxes in first comment. 

Jason

ldpgh, 2017-02-05:
note from the chat
... first try to set some settings on the dialogs in Windows and see if it helps. Some things to check are parent, modal and utility dialog. It may be an easy change. Take a look at the JDialog API docs and see if there is a flag that might help.