openpnp/openpnp #387: GCode driver timeout not configurable

kinnison, 2016-12-22:
Hi,

I am part of a team working on a PnP machine using OpenPNP.  We are using a Smoothie board as our GCode interface, and in order to validate our hardware build, we're running the machine quite slowly while we verify performance and precision characteristics.  Sadly this means that OpenPNP often times out because 5 seconds is not long enough. While I *can* build a local version of OpenPNP with the timeout set higher, it would be awesome if it were a configurable item which could be set in machine.xml.

What do you think?

Daniel.

vonnieda, 2016-12-22:
Hi @kinnison, are you referring to the movement timeout? This is set with the attribute timeout-milliseconds.  You can see it set to 5000 here:

```
<driver class="org.openpnp.machine.reference.driver.GcodeDriver" port-name="/dev/tty.usbmodem1411" baud="115200" flow-control="Off" data-bits="Eight" stop-bits="One" parity="None" set-dtr="false" set-rts="false" units="Millimeters" max-feed-rate="1000" timeout-milliseconds="5000" connect-wait-time-milliseconds="1000">
```

But you can change it to any value you like.

kinnison, 2016-12-22:
Aah I failed to see it in the GUI, should I quit openpnp and edit the XML directly for those?

vonnieda, 2016-12-22:
@kinnison Yes, for now. I'll update this issue to reflect that we should add it to the GUI.

kinnison, 2016-12-22:
I have edited the XML and can confirm that things are now behaving as I would expect.  Yes it would be nice if the GUI could configure timeout, connect-timeout, and feed-rate then :-D

vonnieda, 2016-12-22:
Merry Christmas! https://github.com/openpnp/openpnp/commit/6ce5c9657d6d015bf01e9b1eb7c4c8945d8c3942

commit 6ce5c965 (2016-12-22): Fixes #387: Add GUI controls for setting GcodeDriver units, timeout, connect wait time and max feed rate.
closed 2016-12-22 by vonnieda