openpnp/openpnp #392: Performance Degraded in Recent Release

vonnieda, 2016-12-27:
Users report that OpenPnP is taking more CPU in develop branch and it is better in master branch. At the moment, develop is at a123714f03db998def0ca32e7cc88ce298880f1a and master is at 811175efa6a283c7b708f0b4894f22aaf3c36f4a.

Primary changes between the two are a number of property change listeners, nav view stuff and rearranging the UI.

Discussion in: https://groups.google.com/d/msgid/openpnp/da16baa4-babd-4b84-be66-191d7ac850fb%40googlegroups.com?utm_medium=email&utm_source=footer

vonnieda, 2016-12-27:
Tested in Mac and not seeing significant differences in performance. Master is at about 16% average and Develop is about 18% average. Testing on Windows next.

vonnieda, 2016-12-27:
LA3PNA on IRC reports the same issue in Win 7, 64 bit. The two users on the mailing list are also reporting Win 7, 64 bit.

vonnieda, 2016-12-27:
I was able to reproduce this to some degree in Windows 10. On master the CPU usage was negligible and on develop it was pushing 20%. The bulk of the problem is in the nav view when rendering a new frame from the camera. I suspect that the reason that users saw improved performance when switching to a real camera is the decreased frame rate.

vonnieda, 2016-12-27:
Still not sure this isn't just a Win 7 issue. Even with the ImageCamera running at 24 FPS (instead of the default 10) the CPU usage was staying under 20%. Nowhere near the 60-100% being reported by users.

vonnieda, 2016-12-28:
LA3PNA tested a test build with the nav view disabled and it solved the problem. This narrows the issue down to the nav view.

vonnieda, 2016-12-28:
I pushed a second build which re-enabled the nav view and disabled camera image rendering in the nav view. LA3PNA tested this and said it spiked the CPU again. So it seems to be something related to the nav view but not the cameras.

vonnieda, 2016-12-28:
Looks like resetting config fixed the issue for LA3PNA. This highlighted a bug, but I am not yet sure if it's "the" bug. LA3PNA had a bad camera config, so getWidth() was looping forever and was clearly the cause of the spin. But the users who originally reported this issue said they had no cameras configured. Now waiting on confirmation from them as to whether resetting config fixes it.

In any case, the while loop in getWidth and getHeight needs to be changed / fixed.

vonnieda, 2016-12-28:
Think this has been tracked down to bad camera configs returning null in internalCapture(). I recently (and stupidly) added some while loops to avoid null images, but if the camera is never going to return an image because it has a bad deviceId, for instance, then these just spin forever.

Need to address the core problem, which is that several the camera drivers will return null images from internalCapture(). Need to decide if that is how we want to handle bad configs or if it would be better to just raise an error to the user, or maybe just return a default "bad image" image. Will work on this more tomorrow.

vonnieda, 2016-12-28:
Note to self: We want to handle retries on cameras since that does seem to happen with valid cameras now and then, but we need the retries to bail eventually.

la3pna, 2016-12-28:
Afaik the null images dont last more than microsec, so either triggering a
timer or just retrying 10 times before bailing should fix this?

Thomas.

vonnieda, 2016-12-28:
Change has been pushed to develop. Awaiting results from users.

vonnieda, 2016-12-28:
LA3PNA confirms this fixes the issue.

vonnieda, 2016-12-28:
Now confirmed by several users. Closing.

dzach, 2017-01-08:
Sorry to report late. 
Unfortynately I still get this problem on Linux. With a Philips SPC900NC webcam, which openpnp does not recognize, I get close to 50% CPU load. When I remove it and attach a Philips SPZ5000 webcam, which openpnp does recognize, I get somewhere around 15% CPU load. I have just pulled and run the latest development branch.

My 4 core AMD system [is described here](https://github.com/openpnp/openpnp/issues/395#issuecomment-269798326).

vonnieda, 2017-01-08:
@dzach I don't consider it a problem if this happens on a camera that is not recognized. If someone can submit a fix for that case I'll be happy to have it, but I don't think it's worth spending time on since the camera doesn't work anyway.

dzach, 2017-01-09:
No problem. Scanning through the posts, I saw this remembered that I observed the high cpu load when testing the CCD on Linux and thought it might be worth mentioning that the problem appears on it on Linux too.

dzach, 2017-01-09:
>  I don't consider it a problem if this happens on a camera that is not recognized.
Well, it happens also without a camera connected, as long as there is one configured in machine.xml. If none is configured, or if the one configured is recognized, then cpu load looks normal.

vonnieda, 2017-01-09:
@dzach Ah, okay. That is at least something I should be able to test and reproduce. I'll reopen this ticket and look into it.

markmaker, 2022-02-03:
There seems to be multiple aspects to this:

1. Some CPU load seems _not_ related to OpenPnP at all, but to the operating system and device driver stacks (at least on Windows).  This is easy to misjudge, because CPU load is related to lighting and exposure settings, so improving your lighting might lead to more CPU load. This can now be tested with the Test button of #1053. I have documented the phenomenon [here](https://makr.zone/camera-fps-cpu-load-and-lighting-exposure/519/). AFAIK, there is nothing we can do, I can see the same load in other (non-Java) applications, using the two webcams at the same time (a use case, which one must realize is very rare). 
2. Some CPU load is due to excessive preview camera FPS settings. This has been addressed by providing a suspend-during-tasks options and by Issues & Solutions recommending lower preview FPS and said suspend option.
3. A camera thread bug has been fixed in #1119: sometimes a thread was not properly stopped, when the camera was restarted, so two or more threads would keep capturing images in the background, doubling the load (which happened whenever pressing **Accept** on the camera device settings). 
4. The camera frame broadcasting has been improved, most notably the preview will no longer steal frames from a running tasks, but instead reuse the already captured frames, if available. 
5. Having observed the performance of current versions on both Windows and Linux, I see no performance issues that go beyond the explanations provided in (1) and (2) above. 

All in all, I think it is fair to close this. 

If an unrelated CPU issue is still present, it should be documented in a new issue. 

_Mark

closed 2022-02-03 by markmaker