openpnp/openpnp #395: OpenCvCamera Preferred Height and Width Fails on Linux

vonnieda, 2016-12-30:
User reports that no matter what preferred height and width are set to the camera always returns images at 640x480. Native resolution is 1920x1080. Report is for Debian Jessie. Says that doing the same thing with Python bindings works fine.

See discussion at https://groups.google.com/d/msgid/openpnp/CA%2BQw0jwGyRBWXpvvG%3D%2BY-UqHn_MQvS9%2B%2BvaXOjve99-T%2BCY4vg%40mail.gmail.com?utm_medium=email&utm_source=footer.

I suspect that the version of OpenCV that we're distributing is linked against a camera driver that does not handle property setting properly while his local copy of opencv does. Need to run some tests on a Linux based system.

It would be helpful if others can verify this issue. To do, follow these steps:

1. Set up an OpenCvCamera camera in OpenPnP.
2. Set preferred width and preferred height to something other than the camera's native resolution.
3. Right click the the camera view and select Show Image Info. See if the resolution shown matches the values you set.

cri-s, 2016-12-30:
It works if resolution changes and encoding are changed before opening the camera. I know, Mac is different.
This for 5mpx or less cameras. For higher pixel cameras v4l-dev must be recompiled with increased memory buffer or camera will not work.

cri-s, 2016-12-30:
I know certain libs and cameras work after having opened the capture device, example v4l-dev or was it vid4l-dev,
but it don't work on all cameras.

vonnieda, 2016-12-30:
I have released a test version at http://openpnp.org/test-downloads/ that sets the value both before and after. Please test and see if this works for you.

dzach, 2016-12-30:
I can confirm the problem on a Linux AMD64 machine running Kubuntu 16.04, with a 1280x1024 Philips SPZ5000 Webcam:
```
$ uname -a
Linux desk 4.4.0-36-generic #55-Ubuntu SMP Thu Aug 11 18:01:55 UTC 2016 x86_64 x86_64 x86_64 GNU/Linux

$ dpkg -l | grep libopencv
ii  libopencv-core2.4v5:amd64                       2.4.9.1+dfsg-1.5ubuntu1                           amd64        computer vision core library
ii  libopencv-highgui2.4v5:amd64                    2.4.9.1+dfsg-1.5ubuntu1                           amd64        computer vision High-level GUI and Media I/O library
ii  libopencv-imgproc2.4v5:amd64                    2.4.9.1+dfsg-1.5ubuntu1                           amd64        computer vision Image Processing library

$ dpkg -l | grep v4l2ucp
ii  v4l2ucp                                         2.0.2-4                                           amd64        Video for Linux 2 Universal Control Panel

$ java -version
openjdk version "1.8.0_111"
OpenJDK Runtime Environment (build 1.8.0_111-8u111-b14-2ubuntu0.16.04.2-b14)
OpenJDK 64-Bit Server VM (build 25.111-b14, mixed mode)
```

Could the fact that openpnp cannot recognize a CCD Philips SPC 900NC webcam be of relevance?

I'll try to run the test branch and report back.

vonnieda, 2016-12-30:
Thanks for checking Dimitrios.

The SPC 900NC does not appear to be a UVC device, which means it will not
be recognized. In general, only cameras that follow the UVC standard will
work. These are cameras that the OS should recognize just by plugging in,
without installing any drivers.

Jason

dzach, 2016-12-30:
The SPC 900NC is recognized by the OS without any driver installation, **v4l2ucp** control panel shows all controls,the same way it does for all other webcams, and seems to be able to control the camera. Cheeze and VLC and other multimedia/communications apps that I tried it with do recognize it and work properly. I like this CCD camera for its low light sensitivity and the absense of the rolling shutter effect, despite its mere 640x480 px resolution.

vonnieda, 2016-12-30:
This message seems to shed some light on why the camera is not recognized:
http://nuigroup.com/forums/viewthread/291/

If you can get the camera working with OpenCV and tell me how you did it I
will be happy to make sure it's supported. OpenCV is the limiting factor.
That's how we talk to the cameras. Unfortunately we don't have a native v4l
driver.

dzach, 2016-12-30:
Thanks for giving it a thought Jason. 
You are right about it been peculiar. If I remember correctly I had troubles getting it to work with mjpg_streamer some (long) time ago, when moving from v4l to v4l2. And, I just tried it again, and still, no luck with mjpg_streamer.

nanodude, 2016-12-30:
Jason, I've tested the the version you released and the problem persists. Setting the resolution before or after opening the device has no effect. The only difference is that prior to opening, cap.get returns (0, 0) and after opening it returns (640.0, 480.0). I'm going to try to compile the latest opencv2 jar and use it to see if it works. I'll keep you posted. Thanks!

vonnieda, 2016-12-30:
Thanks for testing Carlos, let me know if you find anything.

nanodude, 2016-12-30:
Quick question: should it be enough to replace the opencv jar file and make the new dynamic opencv java lib known to the OS? I did exactly this (for the lib I set LD_LIBRARY_PATH) but OpenPnP bombs... I may be doing something wrong. Thanks!

vonnieda, 2016-12-30:
Carlos,

As long as you have the same version of OpenCV installed on the machine
then OpenPnP should use that before trying to use it's own. Just make sure
the exact same library filename is available in LD_LIBRARY_PATH. If you are
using the develop version of OpenPnP this will be libopencv_java2413.so.

Jason

nanodude, 2016-12-31:
Well, it seems the problem is that OpenCV doesn't include the nu.pattern thing...

vonnieda, 2016-12-31:
Carlos,

Some explanation is in order: OpenPnP does not use the opencv.jar that is
built by OpenCV. It uses https://github.com/openpnp/opencv, instead. This
is a library originally created by Pattern Consulting which I forked a long
time ago and now maintain. The purpose of this library is to bundle the
opencv.jar along with all of it's dependencies for various platforms. This
is so that I can distribute OpenPnP without users having to build and
install OpenCV externally.

The nu.pattern stuff is the loader for the included binaries. It first
tries to load the library from the OS and then if it's not found it
attempts to load one of the binaries that is included with the library.

So, as long as you have a library in your LD_LIBRARY_PATH with the same
name as the bundled one (libopencv_java2413) it should try to load that
first and use it.

Jason

cri-s, 2016-12-31:
On my laptop and others it works when setting is set before open the
camera. But im forced to use external opencv because otherwise every use of
opencv generate java hard fault core dump. I know it works on annother
Linux distribution. I have commented out the nu.pattern calls and replaced
the openpnp opencv inside pom.XML with locally maven installed opencv.jar .
This just as info.
You should try out webcam instead of opencv camera. Probably it uses opencv
driver too.
For persistent camera mapping, you have to erase the Linux distribution
specific dev part from camera name that appears under /dev/, example
/dev/v4l/by-id/usb-046d_0819_92E84F10-video-index0 index0 part have to be
removed inside machine.XML  that naming is Linux distribution specific.
Probably by-name is used instead of by-id, that don't change the part to
erase. Under GUI, it saves the complete name.
This only in the case resolution settings works
with webcam driver. I don't know. I personally use c compiled shell script
for runas permission reason that recall the camera settings, reload USB
kernel driver, change USB sleep policy in order it recovery from every USB
camera fault seen on some cheap cameras that either have emi problems or
overheating. As you probably understand this sort of shell script is
specific to Linux distribution and installed packages.
I have used this program:
http://www.datsi.fi.upm.es/~frosal/sources/shc-3.8.9b.tgz if you dont have
otherwise success i can help you to setup the camera part.

nanodude, 2016-12-31:
Jason, following your explanantion, I deleted libopencv_java from the jar file (to be sure) and made available my version (via LD_LIBRARY_PATH). Result was the same. There's something strange going on with the java interface. At this point, I don't see what else I can do. Thanks for your help, though!

cri-s, 2016-12-31:
Do you have tested Webcam driver?

dzach, 2017-01-06:
@vonnieda 
> If you can get the camera working with OpenCV and tell me how you did it I will be happy to make sure it's supported. OpenCV is the limiting factor. That's how we talk to the cameras. Unfortunately we don't have a native v4l driver.

I was able to solve the problem with the **Philips SPC 900NC CCD camera** by turning on _WITH_V4L_ and _WITH_LIBV4L_ when running _cmake_. I created this [opencv issue](https://github.com/opencv/opencv/issues/7974), that shows more about the process. 

I don't know whether it is feasible to do the same in openpnp. In the process I learned that one could also use GStramer plugins to pass video/image frames to opencv.

vonnieda, 2017-01-07:
Thanks for checking into this @dzach. I build and distribute OpenCV for OpenPnP, and it looks like the current build did not include V4L. You can see the results from the build here: https://github.com/openpnp/opencv/blob/v2.4.13-0/src/main/resources/nu/pattern/opencv/linux/x86_64/cmake.log#L353

I've added an issue for myself to see if I can include V4L and GStreamer in the next build: https://github.com/openpnp/opencv/issues/13. I will update this ticket once I investigate further.

vonnieda, 2019-02-22:
Closing this as "Won't Fix". Most users have transitioned to OpenPnpCaptureCamera, and I suggest that all other users do so. The OpenCV camera bindings are terrible and unmaintained, so I won't spend any further time updating them.

closed 2019-02-22 by vonnieda