openpnp/openpnp #403: Placement location calculation error on board bottom when using non-corner board origins

vonnieda, 2017-01-07:
1. Create a board that uses a non-corner board origin. In the test case it was near center.
2. Load board into job, select Bottom and set origin. Leave width = 0.
3. Use position camera button to move to a placement. It should position correctly.
4. Set width of board.
5. Use position camera button to move to the same placement. It will position incorrectly.

Discovered when debugging an issue @sp-apertus had. More details to come.

sp-apertus, 2017-01-07:
Actually it should position correctly for the top side (with and without width set), but as soon as you switch to bottom side it will position incorrectly.

vonnieda, 2017-01-07:
@sp-apertus Yes, this issue only affects bottom. Bottom is handled differently than top since everything is flipped and that extra handling is where the bug is.

vonnieda, 2019-02-22:
Note to self, considering removing "New Method" entirely to simplify UI and code: https://groups.google.com/d/msg/openpnp/054jwfjeePY/g-WVQijSAQAJ

Also, an idea: Rendering the placements over the board and having a similar "grab two points" UI to the manual PnP tool would make it very easy to set board origin.

vonnieda, 2020-02-01:
Closing as not a bug: If width is being used on a board bottom the origin needs to be set to the point that is the negative of the width.

closed 2020-02-01 by vonnieda