openpnp/openpnp #405: Add a gcode entry widget for debugging

dzach, 2017-01-10:
A single line entry widget could help enormously debugging when doing setups. Why?

1. Jogging with the mouse or the keyboard is not always the easiest/fastest way to move to a location.
2. There are numerous gcode commands that are used to setup a machine, e.g. defining speeds, offsets, etc, that need direct access to the machine through gcode.
3. Even when doing the setup by editing the configuration files, it's not easy to browse through the verbose log files/screen to match machine replies with each gcode command sent.

The widget could appear at the bottom of the log tab. 
Capability to browse through history of entered commands would also be nice.

vonnieda, 2017-01-10:
This is something that could be added to the GcodeDriver setup panel. It can't be added to the log panel since it is driver specific.

dzach, 2017-01-31:
Well, this is how far my "infinite monkeys" java programming style of copying/changing/pasting existing code can get me with helping here:

[image]

Changed the tab sequence, since usually one sets the serial port only once but hassles with Gcode a lot more. Added a `Console` tab to the GcodeDriver panel.

[image]

The `Send` button at present has its _AbstractAction_ empty.

The capability to interpret macros (possibly words prefixed with an '@', e.g. `@place ID1`) and run Gcode command settings (e.g. the PICK_COMMAND, shown in the screenshot), could help a lot in setting up a machine.

dzach, 2017-01-31:
The above is done with a few lines of code in a file _GcodeDriverConsole.java_
I can attach it or push it, of course.

vonnieda, 2017-01-31:
I like it! I'd suggest first just having the send button call GcodeDriver.sendCommand(String), which just sends a raw string to the port, then later consider adding the macro functionality. I think being able to use @COMMAND_NAME is a good idea but for it to be very useful we'll probably need to provide some assistance to the user to make sure the various variables are supplied.

You'll notice in the CommandType enum I started filling in the variable names available to each command. The intention was that this could be used for exactly something like this someday. So, maybe when the user types "@PICK_COMMAND" the text field would recognize the command and update to something like "@PICK_COMMAND Id=? Name=? VacuumLevelPartOn=? VacuumLevelPartOff=?".

Anyway, I think that should be considered a stretch goal, and just being able to send simple text commands and see the output would be of great benefit.

dzach, 2017-01-31:
So xxx_COMMANDs become generic @macros, and can also be used, some day, in user defined buttons. This is too good to be true.

I'll try the _GcodeDriver.sendCommand(String)_ and see what happens.

vonnieda, 2017-01-31:
Yea, just to expand on that a bit, my plan was that eventually instead of
trying to remember all the variable names, when you are editing Gcodes on
the Gcode panel there would be a third dropdown with variables. When you
select a command it would populate that dropdown with all the variables for
that command so you can just select them to insert them.

dzach, 2017-01-31:
I don't know the keyboard event workings of Java, but maybe an inline <tab> keypress and a popup menu would be easier?

If it were Tcl/Tk I could help a lot more. With Java I feel a total idiot.

Speaking of which: 
`
[ERROR] sendCommand(java.lang.String,long) has protected access in org.openpnp.machine.reference.driver.GcodeDriver
`

vonnieda, 2017-01-31:
Re: keypress: If you mean a special shortcut, I prefer visual over hidden.
A shortcut just means no one will ever know it's there and it won't get
used. A dropdown is visible and people are more likely to go explore it and
see what it does.

Re: error: Change the modifier (protected) on sendCommand to public. I made
it protected because I didn't think people would need to call it outside of
subclasses, but now we do, so it's okay to change it.

dzach, 2017-01-31:
It could be a right click with the mouse. The reason I mention it is because the Gcode console might need it too, if it is to be able to run today's settings commands. 

Re: error: I already did that earlier but ran into more problems:
`
non-static method sendCommand(java.lang.String,long) cannot be referenced from a static context
`
Also changed the method to`static` but that didn't help either. Hence the "infinite monkeys" programming style I mentioned before.

`non-static variable timeoutMilliseconds cannot be referenced from a static context`

Same for `responseQueue`, `portName`,  `output`, `getCommand'.

Jason, this is not helping provide a solution, but rather bothering you withLIFO requests. Didn't mean to.

vonnieda, 2017-01-31:
It's not bothering me. If I can help you learn a little Java and you then commit useful code I have saved myself a lot of time and maybe gained a contributor. A good investment :)

You only need to make the protected -> public change. Nothing else. Then call it like this:

`driver.sendCommand(...)`

driver is defined in GcodeDriverConfigurationWizard, at the top. This is passed in when the wizard is created and represents the driver that you are working on. If you need to use it in a second panel just pass it along when you create that panel.

dzach, 2017-01-31:
All right, that did it. I got back the replies into the text area, though for some long executing commands, like G29, autoleveling, the 5000ms general timeout isn't adequate. Which means that I , either have to parse user input to catch slow returning commands and supply a suitable timeout value, or to check periodically if the controller returned anything new.

[image]

I need to find out if there is an event occurring on serial input. That would make things easier than polling continuously for available controller output.

vonnieda, 2017-01-31:
I'd suggest just adding a second textfield with the timeout. Default it to
5000 and let the user change it if they need to. Send it with each command.
There is a second sendCommand(String, long) that takes the timeout value.

There is no event that comes from the high level driver on data from the
controller, and you have to be careful about what you consume anyway so
that the drive itself doesn't stop getting the responses it needs. I'd
recommend just only showing the replies to the command you send for now.
They will include anything that has been stored up since the last one.

Jason

dzach, 2017-01-31:
Marlin (the controller I'm using right now) sends `busy` messages while doing long commands, but it seems there is some buffering going on somewhere and I'm not receiving them in time. These messages would be sufficient to make the console keep waiting until an `ok` arrives or no more `busy` messages arrive and the command times out.

I've [pushed this  code](https://github.com/dzach/openpnp/commit/5277b901d0eba7a18bdae1b9d9783f1ef67c87db) to my forked repository as is. Have a look if you like. I'll have another go with it and see what more I can do.

vonnieda, 2017-02-01:
Took a look through the code, and it looks good to me! I think that as it stands now it could be merged and close this ticket, unless you want to go the extra distance and add history.

dzach, 2017-02-01:
Yea, let's wrap it up nicely. About 50 entries deep history sounds adequate enough, if not too many. A simple array of strings and a couple of indices, may suffice. That'll be about 12.75kB low utf-8 1 byte chars. I thing I may do it using an array and an index.

vonnieda, 2017-02-01:
This is now fixed by #438, thank you @dzach!

closed 2017-02-01 by vonnieda

dzach, 2017-02-14:
While playing with the gcode console I noticed that OpenPnP's state is messed up when certain gcode commands are sent to the controller, like G0, or G90, G91, G92 commands. I it looks like the mess might not simply be non updated internal coordinates but more. For example, the granular distance slider seems to be influenced as well as the direction of moving ( I can't reproduce it now, but it happen once).  My limited knowledge of OpenPnP and Java do not help me debug this problem. 

I find it pointless to add checks on every command issued by the user, only to keep the machine state updated. We could, instead, add a warning about destroying the state of the machine and add a `reset state` button or add similar default functionality to the `Home` button. 

Should I add a new issue about this?

vonnieda, 2017-02-14:
@dzach There isn't really a good way to reset state, unless the user also specifies the command they want to use to reset state. I'd suggest that we just give a warning that the user should re-home the machine before using it again or restart OpenPnP. The latter would really be the only safe way. So maybe it would be best if the first time the user sends a command with the console we pop a message that says "You should restart OpenPnP before using the machine after using the Gcode Console." or something.

Feel free to file an issue on this if you like.

dzach, 2017-02-14:
OpenPnP is agnostic about the gcode commands the user configures in e.g. HOME_COMMANDS. Maybe a `G28` should be made a special case and do an internal reset?

I'll see if I can add this text inside the widget, faintly colored like HTML5's `placeholder` attribute in an `<INPUT>` element, and have it deleted the moment the user sends in the first command. 

Besides that, I also need to correct history length to 50. It was left by mistake to 5, which I used for testing. I'll do the corrections without opening a new issue.

vonnieda, 2017-02-14:
@dzach Doing an internal reset is hard, because there is not really such a thing in OpenPnP. We could try to parse the G28 and fill in the axis variables from it, but I think that is too much and ignores the fact that people use the driver with non-Gcode based systems. 

I think just giving the user a warning is good enough. The console is really only intended for use during setup, anyway, not during normal use.

dzach, 2017-02-15:
@vonnieda  My lack of knowledge of the way people use the GcodeDriver in non-gcode systems does not allow me to comment more on this, however I would find it reasonable to require these systems to respect gcode commands in a gcode driver, especially G28 for homing to the _machine origin_ and a few others that manipulate the coordinate space.  

Doing a restart is a way of doing a such an internal reset, albeit an ugly one. Had a `@macro` system existed, I would imagine a `@reset_coordinate_space` macro (or `@home`) that would do such an internal and machine reset/sync and could be used by any script in OpenPnP,  gcode included.

No matter what, a warning is a good way to let users know that they are on their own with the gcode console. Maybe a little more comfort could be added in the future.

vonnieda, 2017-02-15:
Agreed. Let's start with a warning and if this comes up again we can revisit.