openpnp/openpnp #406: Error when no dispenser configured

dzach, 2017-01-11:
When there is no dispenser configured, machine cannot proceed to job execution.

vonnieda, 2017-01-11:
What type of job? Please provide steps to reproduce.

If it's a paste job I think it's okay that a dispenser is required to be defined.

dzach, 2017-01-11:
Any type of job. It doesn' execute a single step, no matter what.

**How to reproduce**:

1. Start with a new machine, by deleting the config directory _~/.openpnp_.
2. Proceed as usual. Configure machine, driver as _gcodeDriver_.
3. Run job _pnp-test.job.xml_
4. Verify machine executes job without problems
5. Go to [Solder paste] tab. Delete all entries listed there, for all jobs. Save job.
6. Exit openpnp.
7. Edit _machine.xml_: delete element \<paste-dispensers\> and all its children.
8. Run openpnp again. Go to the [Dispense] tab, under the camera window. Verify there is no dispenser listed there. Go to  [Solder paste] tab. Verify there is nothing listed there.
9. Start the job. Error message:
```
No default paste dispenser available on head H1.
<Try again> <Pause job>
```

vonnieda, 2017-01-11:
@dzach Okay, thanks. I will look into this. I think it's probably a bug in the glue processor. We fixed one in https://github.com/openpnp/openpnp/issues/369 but I think there is probably another.

vonnieda, 2017-01-12:
Okay, the issue is that the glue job processor tries to grab a reference to the default dispenser even if it doesn't have any work to do. We can fix this by deferring the reference until it's actually needed. I'm also going to remove the default glue and paste job processors since these are seldom used and are buggy and not finished.

dzach, 2017-01-12:
Does it mean there will be no paste capability, if the default paste processor is removed? Or one will have to include it manually?

vonnieda, 2017-01-12:
@dzach It will just be disabled by default, and users can enable it by adding the job processors.

commit faffdf3a (2017-01-15): Fixes #406:
* Removed the default glue and paste processors.
* Not having a dispenser will no longer cause an error on pnp jobs.
commit 321f8047 (2017-01-17): Merge branch 'develop' into feature/#399-slotted-feeders

* develop:
  Fixes #417: Nozzle tip name is now editable.
  update attached nozzle tip with machine activity
  show that no Nozzle Tip is attached when null is returned
  nozzle tip name added
  added a light "Orange" to the crosshair color selection
  added five and ten unit ticks
  Fix a couple other confusing uses of the word height WRT #410.
  Making it clearer what "height" and "length" refer to
  Fixes #261 by allowing up to 8 digits of precision for the units per pixels values.
  Fixes #346: Several FeederPanel job tasks were not [... cut]
closed 2017-01-15 by vonnieda