openpnp/openpnp #407: Unable to change config directory with -DconfigDir

dzach, 2017-01-11:
On Linux, I'am unable to run _./openpnp.sh -DconfigDir=/another/config/directory_ and create/use another machine. Openpnp insists on using the default location.

I had a look inside the file _./openpnp.sh_, in the **develop** distribution. It doesn't seem to be passing arguments to the executable *.jar. I also ran the *.jar file on its own, with the _-DconfigDir=/another/config/directory_ option. Neither that made any difference.

vonnieda, 2017-01-11:
The -D command line arguments have to be passed to the JVM directly. 

Try: java -DconfigDir=/another/config/directory -jar target/openpnp-gui-0.0.1-alpha-SNAPSHOT.jar

The order of the parameters matters. The -D has to be before the jar file.

dzach, 2017-01-12:
That did it. Didn't think the order mattered. 
Thank you Jason.

dzach, 2017-01-12:
Changing the following line in script file _openpnp.sh_ makes it work with the command line and the -D option on Linux:

`
java $1 -jar $rootdir/target/openpnp-gui-0.0.1-alpha-SNAPSHOT.jar
`

I prepared a small _patch.diff_ file, but for some reason cannot attach it here.

closed 2017-01-12 by dzach