openpnp/openpnp #411: Strip Feeder Auto Setup Orientation Issue

sp-apertus, 2017-01-15:
I oriented a 0402 tape in my machine as outlined here:
[image]

I then navigated to a "first part"  and the autosetup recognizes it, then the "second part" (below) which the auto setup also recognizes. Then after the auto setup is complete the camera automatically moves to the index 0 part position which often happen to be on the wrong side of the holes - as the autosetup seems to be thinking the tape is oriented differently (reference hole position also indicates this - "next hole has higher Y coordinates").

Screencapturing of the process: https://www.youtube.com/watch?v=8wlNMj2uhqQ

vonnieda, 2017-01-15:
Related to #224. This happens when the wrong holes are identified as the reference holes.

I have some ideas on how to fix this but I have not had time to really flesh them out. The basic idea is something like:
* Instead of having the user click on two parts, have them click on the first part and drag in the direction of travel. This can be the entirety of the setup operation.
* Use the vector they indicated to find the first reference hole by searching both sides of the tape. The user does not always follow the instructions in the graphic above, but we should be able to easily detect that by searching for hole in two spots. 
* If the direction of travel has been indicated incorrectly, just correct it internally by storing an indicator.
* We can also use a similar search process to find the width of the tape. We will need to check both sides each time to make sure we're not seeing the holes from another tape. The closest wins.

dimitarpenev, 2017-01-20:
Hello, I am not in the openpnp but, is it not possible the top camera identifying the direction of the tape looking at it and identifying the circles and rectangles?

vonnieda, 2017-01-20:
@dimitarpenev Yes, that's a good idea. There is definitely more we can do to try to recognize the shape of the tape. The rectangles are tricky because they can be of many different shapes, and they aren't always easy to "see" because of the tape color, but one thing we could do is look for lines that travel along the length of the holes indicating the edge of the tape.

This is one area where I'd really appreciate some help, in the form of code :) I have lots and lots of ideas, but not enough time to implement them! :)

vonnieda, 2017-08-29:
Fixed in #628

closed 2017-08-29 by vonnieda