openpnp/openpnp #412: Feature Request: Switch top vision camera units-per-pixel to pixels-per-unit and make vision system Z height aware

sp-apertus, 2017-01-15:
A cameras viewing frustum (yes indeed a weird word :) ) is basically a pyramid: 
[image]

So the pixel-distances scale strictly linearly if we assume we compensate for any lens distortions (as OpenPnP can do) with the distance of objects from the camera (image sensor to be exact):
[image]

Currently OpenPnP requires all objects that require vision of some sort to be at the same Z-depth.

To make it easier to work with different feeders, different height components, different trays, different PCB thicknesses, etc. it would be very convenient if OpenPnP could interpolate the pixels-per-units based on known Z positions (which we have for all pick up and place locations as defined in feeders, PCBs, etc. anyway).

The calibration would require measuring pixels-per-unit at two known Z heights. The further appart these two heights are the more accurate the measurement should be.

I know this has been discussed before but I saw no issue in the system yet so I created this one so we can keep track of it and make it happen :)

vonnieda, 2017-01-15:
Regarding the "Switch top vision camera units-per-pixel to pixels-per-unit" portion of this: I picked units per pixel kind of randomly when I started writing OpenPnP and it has turned out to be a difficult number to work with over the years. Pixels per unit is much easier to understand more intuitive. I suggested doing this as part of the height chance since the height change will be a fairly extensive change and will affect a lot of the same code. 

A note on the units per pixel change: When this change is made, the old element should be left in place in the config but deprecated. The old value should be copied into the new element if it's null and the old set to null. This will remove it from people's configs over time and after a few months we can remove the old element entirely.

A note on the height change: There are a lot of places in the code that use the units per pixel values directly. We'll need to provide a helper function or something to handle the interpolation and this can probably just live directly on camera and help hide some of that complexity from the rest of the codebase.

sp-apertus, 2017-01-20:
One thing that came to mind recently is that we need to get aware of top vision camera off-angle alignment once we start using the camera to see things in different z-heights.

I will continue thinking about what could be an approach to do this.

BendRocks, 2017-03-03:
I've built several PNP machines in the past, and purchased some too. And they all struggled mightily with Z variations. I've built designs with angled lasers so I could read the location of the dot with the camera. I've built designs with Z probes, so it could sense a switch closure, and I've built designs where the Z nozzle could sense resistance. 

But in the end, I'm currently using a Juki nozzle, they are about $10 each. And they give you about +/- 2mm of z height play so that you can almost completely ignore Z height. You can simply ballpark the Z height of all your components in your head during setup and you'll be fine. 

In other words, most of the Z challenges can be almost completely eliminated by the nozzle choice.

KarlZeilhofer, 2017-07-06:
@BendRocks You might be right for the mechanical alignment of parts, but for vision it is important to have a proper scaling of your pixels.

BendRocks, 2017-07-07:
I've not said anything about pixel scaling. Of course that is needed for proper operation. I was merely pointing out that philosophically there are two approaches to Z height and building PNP machines: Care a lot about Z Axis height accuracy (+/- 0.1mm) and work really, really hard to calibrate heights via vision, touch probes, etc, ==OR== use a spring nozzle and don't worry about Z height accuracy (+/- 1.5mm).  To me, the latter is the way to go (having built both).

markmaker, 2021-04-25:
> The calibration would require measuring pixels-per-unit at two known Z heights.

This part was done through #1112 by @tonyluken.

Whether **Units per Pixel** should be renamed/inverted to **Pixels per Unit** remains open. Personally, I believe it is more understandable to say "one pixel is 0.022mm large" than "there are 45.45 pixels in one millimeter". I know that "dpi" indications are often the other way around, traditionally, but those are often round multiples and have historical rationale from manual typesetting that does not apply here.

tonyluken, 2021-04-25:
I don't see any reason to change from **Units per Pixel** to **Pixels per Unit** just for the sake of changing it.  Note that I have been doing some experiments with calibrating cameras with just a single fiducial.  The results so far look promising.  Just by moving the camera around above a fixed fiducial (or for bottom cameras, moving the fiducial around above the fixed camera) provides enough information to remove lens distortion, calibrate units per pixel (at all z heights), location, rotation, and necessary flips.  It can also correct the perspective distortion caused by the camera axis being non-normal to the machine's x-y plane.  Stay tuned for further developments.

vonnieda, 2021-04-25:
I agree - no need to change this. It was a random thought a long time ago :)

closed 2021-04-25 by vonnieda