openpnp/openpnp #418: DRO location

dzach, 2017-01-17:
I was wondering whether the DRO should change location, now that the GUI has a Multiple Window Style. In a multiple window situation jogging is done in a different window and this may require additional clicks to bring the DRO window to the top.

Maybe something like the following mock-up would be preferable?
[image]

dzach, 2017-01-17:
Irrelevant, but, maybe the copyright note should be updated to 2017?

vonnieda, 2017-01-17:
@dzach I minimized the size and position of the DROs because I feel like they aren't actually all that useful. It's fun to watch the numbers change, but I have found that there's very little reason to refer to them since everything is done from the perspective of the camera.

With that in mind, I don't want to use more vertical space in the jog controls unless there is a very good reason to. Can you convince me that it's important to be able to see the DROs front and center when jogging?

dzach, 2017-01-17:
@vonnieda 

:) I don't fancy at all watching DRO numbers change. My main use of the DRO in openpnp (running demos on the air at present) is mostly as a basic tool for sanity checks. Knowing that the machine does move where it is commanded to, is very basic.

There are times when machine offsets can get very confusing and cause obscure setup problems if they are wrong. For a beginner, setting the camera-nozzle offsets can be an issue, especially if he hasn't RTFM. But more often than not, this is the case, and the little DRO with its "click to show relative" feature saves the day.

Practically speaking, the DRO is as useful as the controls that are associated with it. If you think it is not that useful, then the jog panel is not useful either. But even so, now that the jog panel can be left out of the way in a separate window, why would its own display, aka the DRO, move to another window?

Ideally, I think, a display should be directly associated with the controls that make it change, especially when the controls are manual. E.g. I can't imagine changing the time on my watch and having to take out the mobile phone, open another window and only then get feedback from that screen about the changes I make on the watch.  How often do I change the time on my watch? Very seldom. But when I do, I need the controls there, next to the display.

Finally, one could ask the same question about the usefulness of most numerical entry widgets on the openpnp panels, which get their values by clicking on the camera buttons. Why are all these necessary? Why does one need to know what the coordinates of the Reference and Next Feeder Hole locations are? And why are these even editable?

I suspect GUIs take a lot of time to reach a good level of usefulness. Fitting all essential info and controls on a screen's real estate doesn't seem easy. Hopefully the above arguments dramatized the situation enough to prevent further demotion and ultimate disappearance of the poor little DRO, which still stands in glory in the _stable release_.

I find the work of [Bret Victor](http://worrydream.com/) about interface design very enlightening. [This one](http://worrydream.com/#!/ABriefRantOnTheFutureOfInteractionDesign) has more of an engineer's touch to it.

vonnieda, 2017-01-17:
@dzach Thanks for your comments. I will focus on the most practical part of it, if that's okay:

> Practically speaking, the DRO is as useful as the controls that are associated with it. If you think it is not that useful, then the jog panel is not useful either. But even so, now that the jog [...]

In my mind, the primary task of the jog controls is to move the machine around for targeting with the camera. The jog controls are not tied to the DROs, really, except for the DROs reflecting the current position. I don't use the jog controls with the intention of moving to a specific numeric location, I use them to find an object in the camera view.

The reason the DROs were moved, and minimized in the UI, is that after taking a poll the general opinion was that they really aren't used very often. I don't say they are "never" used, which is why they remain, but they are seldom used.

As for the other editable position fields scattered all about, I would say the most common use case for editing these would be when you have multiple objects you need to set a Z value on. You can easily type in the Z value for all your feeders, for instance. But overall, I agree, they are of limited usefulness. It may be worth looking into whether should be editable at all at some point.

Now, with all the hand waving out the way, I will say that I agree with you, specifically for multi-window mode. In multi-window mode it makes more sense for the DROs to be either on the jog controls window or the camera window. In fact, I've been working with a UX expert and her recommendation was to move them to an overlay on the camera view, and I think that makes a lot of sense. 

For now, I will leave this ticket open and if someone would like to take this on they can. I'll adjust the comments at the top to indicate what should be done.

dzach, 2017-01-17:
Thanks for the explanation Jason. Maybe one other important reason for keeping the DROs together with the jog panel (or the with cameras) is that most people expect them to be there. This makes getting acquainted with the software a lot smoother and getting the job done easier, I believe. 

Actually, the recommendation of the UX expert for a combined camera/jog/DRO panel might be an ideal solution. Together with a set of keyboard shortcuts to bring the various controls up or hide them, and some mechanism for user programmable buttons/keys or an API for connecting with external GUIs it could be a complete solution for the GUI, for now and for the future.

Just my 2 cents.

michalsok, 2017-09-27:
I want to throw my two cents in. I program and operate commercial SMT line every day.
There is an occasional need to use the DRO in real life. It is used mostly to troubleshoot problems, measure by how much to move a part that is asymmetrical and reference is not middle so it places with an offset.
Then the DRO and quick (hot key on the keyboard) to switch to relative "x=0, y=0" is critical.
The machines I use have a DRO in the camera overlay and when relative mode is entered it adds a new line with relative XY=0.000 bellow an absolut DRO redout.
I hope this helps. (or adds to confusion)
Mike S.

wholder, 2023-08-03:
I implemented a similar Jog controller for my [LaserCut project](https://github.com/wholder/LaserCut/tree/master), but with addition of X/Y diagonal controls, like this:

[image]

The relevant code is in the file GRBLBase.java.  Feel free to use any code that might be useful, such as the [Arrow class](https://github.com/wholder/LaserCut/blob/master/src/GRBLBase.java#L619) that generates the triangular button graphics.