openpnp/openpnp #419: New Scripting Events: Start/End of job and board

vonnieda, 2017-01-17:
Four new scripting events for start/end of job and board. These will allow people to script things such as conveyers. Suggested by @issus.

vonnieda, 2017-01-27:
@sp-apertus Has created an event wishlist at https://docs.google.com/document/d/1NuC7qGcUZLByHA-fM5qOU7MV9SmNymWq9lGdNJr2WJk/edit#heading=h.wk3qf4uviszt

markmaker, 2021-07-02:
I think these are all done (and many more).

https://github.com/openpnp/openpnp/blob/a9773de4b918d17b43ca780fa428e2571727f506/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L278

https://github.com/openpnp/openpnp/blob/a9773de4b918d17b43ca780fa428e2571727f506/src/main/java/org/openpnp/machine/reference/ReferencePnpJobProcessor.java#L1076

I don't think there can be a "board" event, the job is planned across all the boards, right? 

True automatic conveyor support, in the sense that the same job will be started again and again, as long as the conveyor senses a new board arriving, will require more logic and this seems beyond this issue. 

For the other ones listed by @sp-apertus, I believe all are there (in the sense of today's evolved implementation). 

See also https://github.com/openpnp/openpnp/search?q=.getScripting%28%29&type=code

_Mark

closed 2022-02-03 by markmaker