openpnp/openpnp #420: Wiki Image Upload

sp-apertus, 2017-01-18:
[image]

vonnieda, 2017-01-18:
@sp-apertus If you close this are you still able to comment? I haven't tested that. I like these image upload issues to remain closed, if possible, so they don't clutter the work list.

sp-apertus, 2017-01-18:
works !

closed 2017-01-18 by sp-apertus

vonnieda, 2017-01-18:
Excellent, thanks!

sp-apertus, 2017-02-11:
[image]

sp-apertus, 2017-02-11:
[image]

sp-apertus, 2018-05-22:
[image]

sp-apertus, 2018-05-22:
[image]