openpnp/openpnp #421: Icon consistancy

dzach, 2017-01-23:
On most cases the _plus sign_ icon means `Create a new ...`, as indicated by the help popup on the button. The only exception I noticed (which always confuses me) is its use in the `Job` panel, where the popup says `Add an existing board...`. Here are some example screen shots:

In `Packages`:

[image]

In `Solder Paste`:

[image]

However, in `Jobs` a `pencil` has replaced the meaning `create a new` of the `plus sign` icon, which now means `add an existing` :

[image]

[image]

Is there a reason for that, or it's just an oversight?

vonnieda, 2017-01-23:
@dzach The thinking was more that the plus sign is the way to add something to the list. In many cases the only way to add to the list is to create something new, while in the job panel there are two ways to "add" something - one of which adds existing and one of which creates new.

I find it confusing, too, and I put it there. I'm open to suggestions on how to improve it.

dzach, 2017-01-23:
In this case, `add an existing board` is equivalent to `import a board`. My suggestion would be to use an `import` icon, which is, usually, an arrow pointing down or into a circle or rectangle. The `plus sign` icon then keeps its `create a new ...` meaning.

Import icon example:

https://d30y9cdsu7xlg0.cloudfront.net/png/65557-200.png

vonnieda, 2017-01-23:
Ah, in looking at the Material Design icons (which is what most of the
OpenPnP icons are) I see that the pencil is labeled "create" and the plus
sign is labeled "add", so that is probably why I did it that way. It's been
several years...

Anyway, if your suggestion is that we change the New Board action to use
the plus icon and the Add Board action to use the Import icon, I am fine
with that change. We'll need to find a Material Design icon that fits, or
another free icon that is similar in style.

The Material Design icons are at https://material.io/icons/. Do you feel
any of those fit? Also, keep in mind that we currently use an icon that
looks a lot like that for nozzle load and unload, and I would fine changing
those as well, as they are not particularly clear.

Jason

vonnieda, 2017-01-24:
I see there are more community contributed material design style icons at https://materialdesignicons.com/. I like https://materialdesignicons.com/icon/file-import for import. What do you think?

dzach, 2017-01-24:
Thanks for the links Jason.
I believe the icons from Material Design and all the metrics used there for screens etc are exceptional work. Here is my suggestion for the `Job` panel:

[image]

This is the light background icon from [next to the one in your link above](https://materialdesignicons.com/icon/file-import).

I'll have a look at the nozzle load and unload icons, if I find them :(  and see if I can come up with a suggestion for those too.

Dimitrios

dzach, 2017-01-24:
Ok, how about this?

[image]

vonnieda, 2017-01-24:
@dzach I don't like the import icon you mentioned above. To me it looks like an exit sign and it doesn't make me think of importing or adding a file.

I do like the mock ups of the nozzle change icons. I think that's an improvement over what we have now. 

Perhaps we should look at this from a different perspective. The plus sign is widely known as "Add", not "Create" and in almost all cases we are using it as "Create". Perhaps we should keep the plus sign as "Add" and change the other places we are using it to a "Create" icon.

In other words, we could keep the plus sign for add board and either keep the pencil as create or pick a new create icon. Then we'd change the places that erroneously use the add icon to use the new create icon. This would include Parts, Packages, etc. 

What do you think?

dzach, 2017-01-24:
I think what is significant in `Add an existing board` is the `existing` part. Otherwise `Add` and `Create` seem to have similar meaning. I find the `plus` icon very suitable for what it represents now, except for the case of importing an existing board. I agree with the "exit sign". How about this then?

[image]

dzach, 2017-01-24:
Maybe I spoke too early. My limited experience with OpenPnP might be useful for a fresh look to things, but I'm probably missing a number of points.

Usually the `Create' part is the icon of an empty document with a star, a 'plus' or some kind of glitter denoting the 'newness' of the matter. 

Let me see if I can find a proper `Create` icon, and I'll come back to this.

dzach, 2017-01-24:
Something like the following maybe?

[image]

vonnieda, 2017-01-24:
@dzach I think that looks very good. Would you propose that we use that new icon (the create one, on the left) for all the places we currently use the plus icon, like adding a part, package, etc? Or do you think this would be the only change?

dzach, 2017-01-24:
I see that Google uses the `plus` icon as a generic create/add content icon, when there is no ambiguity about the intent. If there is any conflict in the intent, then that may be an indication that a separate icon should be created. 

It all depends on the context. I had no problem using the plus sign icon as it stands now, except for the `Job` panel, where the pencil, usually standing for `edit`, confused me. So I'd prefer to take the conservative path and leave the other icons as they stand now, and only change the problematic one.

Looking at the other OpenPnP icons, I now see that the `nozzle unload` and `nozzle load` icons I proposed above are better suited for the (non-existent  `place` and the) `pick` command respectively, where the existing  icon (usually meaning `upload` or `export`) serves nicely. 

Since I've given very little thought to all of these, it may be wise to let it soak for a while and see if we can come up with some better ideas.

vonnieda, 2017-01-24:
@dzach This sounds like good advice. I agree, the pencil puts me in the mind of "edit" as well. I think changing just the problematic one is a good first step.

I am going to try to find some apps that have similar functionality, where you can either add an existing object or create a new one, and look for some visual cues there. It could be that we could resolve this just be changing how the add/create functionality works. Just thinking outloud, but what if we removed the pencil / create action entirely and just had the add button. Clicking it would open the file dialog. If you select an existing file it's added, if you type in a new filename one is created.

For reference, you can see all the icons here: https://github.com/openpnp/openpnp/tree/develop/src/main/resources/icons

If you click on them Github will render them so you can see what they look like.

vonnieda, 2017-01-24:
@dzach How about this?

[image]

The confusing create button is removed, and when you click the add button you get a menu asking what you want to do.

I really like this a lot.

vonnieda, 2017-01-24:
@dzach I've pushed a branch that has the above change in it, so please have a look and see what you think. I think this is a big improvement. You can download binaries from http://openpnp.org/test-downloads/ or source from https://github.com/openpnp/openpnp/tree/test.

dzach, 2017-01-24:
@vonnieda I was ready to say that it might be better to avoid adding one more click to the interface, but then I tried the new menu and, boy, do I like it! I think this solves the problem with the confusing `edit` pencil and keeps the generic `add` icon in place. 

For the rest, I propose to add three visual clues to the icons, denoting the nozzle tip, the nozzle tip holding fixture, and the tape/bed of the machine, like these:

[image] Nozzle tip visual clue

[image] Tape & bed visual  clue

[image] Nozzle tip fixture visual  clue

Then we can have icons like these:

[image] Load nozzle tip
[image] Unload nozzle tip
[image] Add tip
[image] Remove tip

but also keep consistency with icons like:

[image] Pick (notice the black visual clue for the tape, used also in the feeder icon)
[image] Place

What do you think? It's by no means complete, but it could provide some base lines for new icons.

dzach, 2017-01-24:
I have all the above in the local openpnp/src/main/resources/icons/ directory of the 'test' branch. I can push these changes to github, if you think it's worth to give them a try.

vonnieda, 2017-01-24:
@dzach I think these look great. I'd love to see them in action. Can you push the branch up?

dzach, 2017-01-24:
I'm trying `git push` but I'm getting the error

`remote: Permission to openpnp/openpnp.git denied`

Obviously, I don't have push permission. Maybe then I should fork the repository and continue that way.

vonnieda, 2017-01-24:
@dzach Yes, please fork, add your changes and then create a pull request. If you aren't familiar with the process, check out https://github.com/openpnp/openpnp/wiki/Your-First-Pull-Request.

Also, you can use the develop branch. The test branch is temporary and was just to show you the changes I made. I'll be merging those changes into the develop branch.

dzach, 2017-01-24:
Will do right away.

WRT the icons, I make here a note to myself to keep in mind that using colors alone as visual clues is not a good practice, since a not insignificant percentage of people have some issues with color blindness. 

So, depending on style, it's good to have some shape differentiation together with a different color, e.g. red color and 'X' denoting removal, or green and clue position to denote loading of a nozzle e.t.c.

vonnieda, 2017-01-24:
That is a very good point. Even after all these years I have to think for a
second before I hit the capture camera / position camera or capture nozzle
/ position nozzle buttons since the only difference is color. I appreciate
any and all help in this area! :)

vonnieda, 2017-01-24:
@dzach Are you satisfied with the changes? Can we close this ticket?

dzach, 2017-01-24:
Yes, especially with the menu for adding a board.
However, I'm still working on some icons for the position of the camera and nozzle buttons. 
What to do?

vonnieda, 2017-01-24:
@dzach If you are still working on icons we'll leave this open. I'll change the title just a bit to be a bit more clear.

dzach, 2017-01-24:
Ok. Here are some crafted icons for the tip operations, based on the visual clues described above. Just an idea which might lead somewhere, some day.

This one looks like a Zuki-504, or something:
[image]

A fictitious generic type of nozzle tip.
[image]

Not many curves, actually ... none! I hope they don't look too old-fashioned .

I'll follow up with the ones for the camera, and then call it a day.

dzach, 2017-01-24:
So, here are the two icons for the `position camera/nozzle` buttons. 

The idea is that the blue color represents the numeric coordinates for the camera/nozzle, and when the relevant button is pressed, these coordinates are captured and transfered to the entry widgets next to the buttons.

In the existing red colored icons, the red color represents the movement of the camera/nozzle to the coordinates represented by the blue color. Let the blue color be the target towards which the camera will move once the button is pressed. The **target**:

[image] Move camera towards the target coordinates

[image] Move the nozzle towards the target coordinates

I love simplicity, but simplicity is hard to achieve. 

Comments and ideas are very welcome!

vonnieda, 2017-01-24:
@dzach First, I just want to say THANK YOU for doing this work. I love what you are doing and appreciate the help!

I really like the Juki style nozzle tip icons. Especially with the visual cue you've designed for the changer, the gap between the nozzle tip and the nozzle holder makes that even more clear. It seems like a good differentiator between nozzle operations and nozzle tip operations would be greying out the top or bottom half of the icon.

I *love* the position camera/nozzle icons. I want to start using them immediately! :) Very excited to see what you come up with for the opposite side of those - for capture.

Thanks,
Jason

dzach, 2017-01-25:
One more. This is for the yellow one in the `Nozzle Tip Changer` panel, that _positions the tool over the center of the location without first moving to safe-Z_.

[image]

The straight line through the target supposedly hints at the _without first moving to safe-Z_ part.

vonnieda, 2017-01-25:
Looks good to me. I can think of it as the line symbolizing the machine
bed, which you are likely to crash into by using that button :)

Jason

dzach, 2017-01-25:
Hey, thanks for loving them, it is very encouraging to hear it! 
These icons are a very small payback to this wonderful open source project you've created and maintain.

The position camera/nozzle icons are based on the "meaning" and the colors of their capture counterparts. Didn't plan to change those, because I like very much their simplicity and power to convey their meaning: Blue straight camera means camera coords, blue straight nozzle means nozzle coords throughout the application.

I see that the new icons have discrepancies in their positions within the buttons. For some reason my installation of Inkscape has a problem with the construction guidelines and I may need to change computer to do some maintenance work to the icons. Will also try to unify the position of the side symbols `plus`, `X` in all icons. The devil is in the detail.

I'll create a branch named 'icons' and put the bulk of new icons there. 
Will follow that with a new pull request.

vonnieda, 2017-01-25:
Ah, that makes complete sense now that you mention it! Very clever! :)

Going to look at the pull request now.

dzach, 2017-01-28:
A new icon, _feeder.svg_, just in case it is needed:

[image]

And a replacement to  _feeder-edit.svg_:

[image]

dzach, 2017-01-28:
I have now corrected the problem I had with Inkscape and, while doing that, I realized that a number of issues need to be dealt with in a more systematic and organized way. 

This is the current palette used in openpnp icons (all light colors, except yellow, were added with the new icons) :
[image]

and this is the palette when converted to gray:
[image]

The luminosities (that is, the _luma_, as Wikipedia corrects me and everybody else) of the yellow and gray colors as they are rendered in the icons, tell that the gray I used in some icons gives worse contrast than the yellow, which is pretty obvious: 
[image]
Comparing the monochrome version of the two feeder-edit icons, the difference also is obvious:
[image]

Fixing such issues may result in a better set of icons for openpnp in the future.

vonnieda, 2017-01-28:
Nice work @dzach. I agree that the yellow is more clear. Would you like to add these new icons to the project now, or are you intending to go forward with a bigger rework?

dzach, 2017-01-29:
I can push the feeder ones above, but they need some changes in the code too:

[image]

I believe these icons should be changed to the new one shown above, _feeder.svg_  like this:

[image]

The _feeder-edit.svg_ should remain in the pick and place panel:

[image]

I've tried these changes in the code, and it passes the tests. Should I push?

There is more work to be done with the icons. As I read somewhere, 

> The icons. They will never be done.

:)

vonnieda, 2017-01-29:
@dzach Yes, please do!

I think I made that "never be done" comment when I was trying to design the first set of icons for OpenPnP. Not being a designer of any form... it took me a long, long time :)

dzach, 2017-01-29:
I spent quite some time trying to align the shapes in the icons. No matter what I do, even when everything sits perfectly on pixel boundaries, the circles come out distorted and asymmetrical:
[image]

So maybe its time to increase the size of the icons to 32x32 and see how that comes out.

Pushed the new icons. See #434

vonnieda, 2017-01-29:
I took a quick look at the position-nozzle icon and noticed it's not quite pixel aligned. Maybe this is a quirk of Inkscape?

[image]

Also, when I was playing around with this, I noticed that using integer sizes for the objects helped a lot too. 

In any case, I think the icons look great so don't feel like you have to keep working at it. I'm very happy with how they are now.

dzach, 2017-01-29:
Inkscape does alter the dimensions some times (I haven't figured exactly what it tries to do) and I have to repeatedly correct them. But look what Inkscape itself thinks of its shapes:

[image]

And then look what openpnp thinks of Inkscape's svg files:

[image]

And this is how the alignment looks for the image above
[image]

And finally, this is that the svg file has inside:
[image]

An open can of worms.

dzach, 2017-01-29:
... and then, the new, v0.92, Inkscape informs me about this:

[image]

Let's carry on with openpnp.

vonnieda, 2017-01-29:
I went ahead and added a little icon viewer. Not sure if it will be useful
or not, but it was 15 minutes of work to find out. In
org.openpnp.util.IconViewer. Just run that, change the directory name in
the text field and hit Refresh. It will display all the svg icons in the
given directory at 24x24 and has a checkbox to turn them gray.

Jason

dzach, 2017-01-29:
@vonnieda  you are talking to a java illiterate here:

src/main/java/org/openpnp/gui/support/SvgIcon.java:25: error: package org.apache.batik.transcoder does not exist

Downloaded batik but that's just a bunch of jar files. Spent more than an hour trying to figure out how to make these available to java and openpnp but had no luck so far.

vonnieda, 2017-01-29:
Sorry @dzach, I should have explained better. Do you have the OpenPnP code in Eclipse? That's the easiest way to use this. If so, just look for org.openpnp.util.IconViewer in the Package Explorer or Java View, right click and choose Run As... Java Application. Should pop the window right up.

If you aren't using Eclipse, let me know how you've been working on the code and I will assist.

dzach, 2017-01-29:
The old fashion way: Use a text editor and then 
`mvn package`
`./openpnp.sh`

or, in this case:

`javac -cp /home/dz/develop/openpnp/src/main/java  IconViewer.java`

It seems to be able to find the java classes but not Batik

vonnieda, 2017-01-29:
Try this:
```
mvn package
java -cp target/openpnp-gui-0.0.1-alpha-SNAPSHOT.jar org.openpnp.util.IconViewer
```

The `mvn package` process adds metadata to the openpnp jar that tells it where to find all the required libraries. At the end of the build they are all copied into target/lib and then the openpnp jar has a reference to lib/libraryname.jar.

dzach, 2017-01-29:
Nice!
Thank you!

dzach, 2017-01-30:
I added a text field next to the directory one, to change the size of the icons:

[image] 24x24px

[image] 32x32px

So, I definitely prefer 32x32px.

Now I went ahead and changed the size in function getIcon in Icons.java to 32 and:

[image]

[image]

I prefer this so much more, I'm keeping this version of openpnp with the 32px icons to work with. 

Your IconView program offers a nice overview of all icons and has already helped identify some details I've missed.

dzach, 2017-01-30:
I am very much puzzled by the rendering of the icons by Batik. Here are some renderings I got from various programs:

[image] Firefox

[image] Konqueror

[image] Gimp

[image] Batik

Of all, only Batik shows a problem with the symmetry of the icon, the rest, though render the same file differently do not show any problem with the shapes. That's very strange.

vonnieda, 2017-01-30:
That is very strange. I'll do a little research and see if there are any
options I should be passing to Batik for rendering.

vonnieda, 2017-01-31:
Alright, did a little research and didn't come up with much, but here is what I found:

* Updating Batik to latest didn't improve anything.
* Loading the position-nozzle.svg icon into Graphic and then re-exporting it didn't improve it, but it did make a huge change in the contents of the svg. The Inkscape one is 152 lines and the Graphic one is 10. I suspect this is because Inkscape treats svg as it's native format, so it contains a lot more metadata.
* Fixing up a couple alignment issues didn't improve anything. Two coordinates were not on quarter pixel borders so I fixed that, but it didn't help.
* Changing *everything* to integer coordinates and sizes, including positions, dimensions and strokes, removes the distortion but then there are some positioning oddities. In Graphic everything is nicely aligned, but then in Batik the nozzle looks offset. 
[image]

So, not sure what to take from this except that maybe Batik is broken, or maybe the icon size is just too small.

One thing we could consider is going back to using PNG icons in the program while keeping the source images in SVG. It would be easy enough to export them all.

Another option, as you pointed out, is that going up a size or two on icons. I quickly tested at 32x32 and the icons are certainly easier to see and understand at that resolution, but it didn't seem to fix the distortion issue.

dzach, 2017-01-31:
No, the distortion is still there, even with the 48px icons that I've tried.
The stand alone Batik does produce the distortion too.
Inkscape does NOT provide a way to have only integer coordinates. There is a plugin that does something with alignment of pixels, but it is not making the coords integers.

Inkscape does use the svg file to store its own, non svg, data, like guideline positions and style e.t.c. There is a "plain SVG" save mode but I didn't use that as I'd loose all the setups of the guidelines for the individual icons. I can do that, but then I will have to keep a different directory for the originals.

I started using the local 32px version of icons and the problem is far less pronounced than with the 24px one, though still there. Openpnp also feels ... more mature with the 32px icons. There may be a need to shrink white space in the Jog panel a little. 

If there is any thought about changing the size of the icons to 32px, then this may be the solution. Otherwise, I can't think of anything else other than reverting to using images.

vonnieda, 2017-01-31:
I've filed an issue for Batik on this https://issues.apache.org/jira/browse/BATIK-1175

vonnieda, 2017-01-31:
Regarding the larger icons. I'm willing to test it out. We can post screentshots to the mailing list and see what people think. My personal feeling is that it makes the app look less modern; more like an older Windows app. That's just my personal opinion though and has no basis in objective reality.

dzach, 2017-01-31:
That was my feeling too until I started using a touch screen with my DIY milling machine. Suddenly I became interested in Material Design and what makes larger icons look more modern... :)  Haven't found an answer, other than simplicity and abstraction of symbols, yet.

Could make icon size a user option, maybe a sub menu in the window menu. 

Let's see what the outcome is from that issue report for Batik.

dzach, 2017-02-05:
If Batik is used for all display rendering, then there may be more problems:

[image]

This is the cross-hair from the down camera in openpnp. Doesn't look much like a circle...

[image]
Here with a threshold applied.

vonnieda, 2017-02-05:
Batik is only used for icon and rendering, nothing else.

vonnieda, 2017-02-05:
-and, mobile so I can't edit. To clarify: batik is only used for icon rendering.

dzach, 2017-02-05:
So, if this circle is not drawn by Batik, could the source of the problem be with Java?

vonnieda, 2017-02-05:
@dzach I misunderstood. I thought you were creating a demonstration, I didn't realize you are saying this is from OpenPnP. I will take a look and see if this happens on my system.

[6 more replies]

closed 2017-03-21 by vonnieda