openpnp/openpnp #422: Machine Setup Wizard

vonnieda, 2017-01-24:
Create a first time setup wizard that will help people configure their machine. The goal of the wizard would be to ask a series of questions for the purpose of setting up the correct number of nozzles, nozzle tips, axes, cameras, gcodes, axis transforms, etc. It would not need to go into the deep details of each item. The goal is just to get a basic config up and running to avoid the many FAQs that come up on the mailing list.

The wizard should live under a menu item and can be run at any time. It should include a warning that the user's config will be completely replaced.

* [ ] Add a new Guided Configuration menu item.
* [ ] Menu item should pop a dialog telling user their config will be overwritten. They can cancel and exit to back up.
* [ ] Dialog with standard wizard style controls (Next, Previous, Cancel) will pop up. Main panel will be used to ask questions.
* [ ] Each question will appear on a panel after clicking next. 
* [ ] Panels / Questions:
    * [ ] "Welcome to setup. This will overwrite your config. Exit to backup. Next to continue."
    * [ ] "What type of controller are you using?" List of common controllers, including "Other Gcode", "Other Non-Gcode"
        * Should include at least Smoothie, Grbl, TinyG, MKS Base (Smoothie)
        * If known, or "Other Gcode", set the machine driver to GcodeDriver. Fill in what we know about the controller, particularly the COMMAND_COMPLETE_REGEX and MOVE_COMPLETE_REGEX.
        * If Other Non-Gcode, bounce them to the mailing list or Wiki. 
    * [ ] "How many nozzles does the machine have?" Dropdown with numbers.
        * Clear the nozzles list, add number of nozzles with easy IDs, add a nozzle tip to each.
    * [ ] "How do the Z axes work? Show options for cam, inverting, independent.
    * [ ] "Is there a down looking camera?" Yes or No
        * Create an OpenCvCamera on the head.
    * [ ] "Is there an up looking camera?" Yes or No
        * Create an OpenCvCamera on the machine.
    * [ ] Configure the GcodeDriver axes based on previous answers.
    * More to come...

Even though the script above says what changes should happen after each step, we should really just store all the answers and make all the changes at once, at the end, so that we don't end up with a partial config.

markmaker, 2021-04-25:
While a different, more flexible i.e. less linear and iterative approach was chosen, this is mostly what [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions) does. 

However, some points are still missing.

vonnieda, 2021-04-25:
> While a different, more flexible i.e. less linear and iterative approach was chosen, this is mostly what [Issues & Solutions](https://github.com/openpnp/openpnp/wiki/Issues-and-Solutions) does.
> 
> However, some points are still missing.

I think this issue as written can be closed. I agree that I&S solves a lot of it, and what it doesn't solve now can be solved by adding more I&S entries.

closed 2021-04-25 by vonnieda