openpnp/openpnp #427: ReferenceSlotAutoFeeder Bugs

vonnieda, 2017-01-25:
Two bugs identified by Glen English:

* [ ] Sometimes goes into a spin loop sucking up CPU. Seems to happen when adding multiple slots, or possibly clicking into the slot name, or possibly scrolling through them rapidly.
* [ ] Slot name gets corrupted if you double click it multiple times. This is due to how we override the name to include the feeder. Need to add a table cell renderer that shows the combined value but make sure when editing we only edit the actual name.

And some improvements:
* [ ] Change the ID prefix for slot feeders to something other than FDR. Maybe SLF.
* [ ] Change the ID prefix for slots to SLOT.

vonnieda, 2017-01-25:
Spin loop error is due to Configuration.get().getBus().post(new FeederSelectedEvent(feeder)); call. Still determining why.

vonnieda, 2017-01-25:
Looks like it's a loop in FeedersPanel.feederSelected, which I suppose makes sense.

closed 2017-01-25 by vonnieda

glenenglishgithub, 2017-01-25:
OK,  I concur that all seems to be bahaving itself now, thanks Jason !

commit 6fa7b040 (2017-01-25): Fixes #427: FeederPanel now ignores feeder selected events that it produces. These were causing an infinite loop. Probably need to fix it everywhere.