openpnp/openpnp #430: ReferenceDriver base class

visualecho1, 2017-01-26:
Would a base class for ReferenceDriver be useful?  For the GcodeDriver, it would fit between AbstractSerialPortDriver and the ReferenceDriver interface.  For other drivers, it would be an abstract of ReferenceDriver.

Many of the drivers have stubs where common functions are not applicable to other drivers, this would provide a common place to provide virtual stubs for otherwise unuseful methods.  This will shorten and clean up the driver code, and fare well for future enhancements...

...such as I need to call a driver method before and after loading or unloading a nozzle using a LitePlacer machine with the upgrade Samsung CP40 inline-6 magnetic nozzle rack.  This will allow deactivating the Z-axis min switch temporarily so the nozzle can be mashed down nice and tight.

Right now to do this, I have to add four stub functions
* pre_load_nozzletipchange()
* post_load_nozzletipchange()
* pre_unload_nozzletipchange()
* post_unload_nozzletipchange()
to AbstractSerialPortDriver and the other drivers which do not extend it.  It makes a mess.  I'd like to work this out, but I can think of a couple of reasons why it might not be a good idea.  It's completely possible that there's a feature in Java that I don't know about that makes this a very bad idea.  Comments, please?

vonnieda, 2017-01-26:
@visualecho1 Please see the comments in https://github.com/openpnp/openpnp/issues/354.

Instead of adding more methods to the driver, I'd prefer to use the Actuator system for this. The Actuator system is designed for driving arbitrary hardware devices in a driver agnostic way. The way your need can be implemented is:

1. Modify ReferenceNozzleTip and it's wizard to include an optional actuator name and value at each move, along with a boolean as to whether it should be fired before or after the move.
2. Define an actuator on the machine for the z min switch.
3. Use GcodeDriver to specify the appropriate outputs for the actuator.

Over time my goal is to make the driver interface smaller, not larger. I feel that the driver interface is basically "complete" at this point and in any cases I can currently think of we should use actuators for additions.

The primary benefit of using actuators is that it doesn't limit the functionality at the driver level. As soon as you add pre_load_nozzletipchange(), for instance, someone will come along and say e.g. "Oh, but I also need to fire a second pin during the second movement". Since the user can define an arbitrary number of actuators, this makes it easy to add this kind of functionality without having to add code to make sure it's supported in every driver.

Note that the idea I've described above will be helpful for others, too. There are people who are starting to use changers that require the rack to be locked and unlocked before inserting and removing. By using actuators they will accomplish this as well.

You can take a look at the code for ReferenceAutoFeeder and it's wizard to see how I typically handle actuator setup for arbitrary devices.

visualecho1, 2017-01-26:
Thank you, I'll look at this!

closed 2017-03-21 by vonnieda