openpnp/openpnp #432: A 'Place' button?

dzach, 2017-01-28:
Just got the solenoid valves and my first pick and place of a part, with my all-in-one 3D printer,  is now a fact. Thank you OpenPnP!

Now, when setting up the machine, I use the 'Pick' button located in the 'Feeder' panel, as suggested by the manual. It picks up the component all right, but then the only choice that I have, is to discard the part. And then, pick them up by hand from the discard bin and put them back into the tape pocket, again by hand.

It would be very helpful to have a 'Place' button next to the 'Pick' one in the 'Feeders' panel, and maybe the same pair of 'Pick' and 'Place' in the 'Pick and Place' panel. With this set of icons in the 'Pick and place' panel one can test an individual pick and place operation for any part, especially for the ones that are prone to fail and need fine adjustments of parameters. Something like this:

[image]

I've noticed that the 'Place' icon is now included in _Icons.java_. Does this mean there is something coming up in this area?

vonnieda, 2017-01-28:
I think adding the pick and place icons to the pick and place panel is a
great idea. For the feeder place one, how would you suggest that worked?
Would it place the part back in the feeder?

dzach, 2017-01-29:
Yes, that's what I thought, use it to place the part back in the feeder, because I got tired of putting back manually the parts that got dropped to the bin after a pick and discard.

Here are some icons to aid to the visualization of the proposal:
[image] Place the part to the indicated location, i.e. where a red nozzle would go. If the icon appears on the feeder panel, then the location is the current feed position, i.e. the part gets placed back in the feeder pocket/tray e.t.c. If the icon appears on the pick and place panel, then the place location is on the board, where a red nozzle icon wound put it. 
[image] A single normal pick-n-place operation.
[image] A pick and place, in place. This can work either by defining a fake feeder (e.g. an AutoFeeder) with a feed location somewhere on the bed, or the current nozzle location, depending on the panel where the icon appears.

I've modified the new 'place' icon and made the base bar blue, to imply that placing is done to the indicated/intended location. The green bar didn't make much sense.

So, in essence, these are the meanings of the colors so far:

[image] Location
[image]
[image] Movement of camera or nozzle
[image]
[image] Fiducial/component
[image] 
[image] Operation, other than moving the camera/nozzle
[image]
[image] Used for shapes that are indifferent to the operation
[image] Feeder/tape

vonnieda, 2017-01-29:
I like the place idea, and the pick and place idea. I'm not sure about the pick and place in place one. I feel like since we already have the pick button and will add the place button right next to it there's no need for another button when you can just click the one and then the other. In addition, I think this is something that is unlikely to be used very much. I'm open to discussion on that though. If you can think of a reason why it would be useful let me know. Perhaps I am just not seeing it.

dzach, 2017-01-29:
My proposals are biased towards the things I'm doing right now, since my experience with openpnp is very limited. I haven't yet run any real job, not to mention a production run (LOL!). So the pick-n-place in-place is what I'm doing right now trying to find the best settings for various parts and needle-nozzles. To do it, I have to create fake feeders with their feed location the same with the location of the placing on the pcb, and then repeat this for the number of times I want. Not nice. I'd like to test my needle-nozzles say, 100 times on picking a single component type, just to see how much reliable a configuration might be, but I don't want to replace 100 x 10k 0603 resistors in their tape. I'd rather design icons for openpnp :P

I spent a little time trying to find the place in the code where I could wet my feet into the troubled waters of Java, but I guess I didn't try that hard. On the other hand, maybe putting all these ideas in openpnp will render the UI cluttered and unfriendly. Or maybe a special 'Testing' panel can be created, where one might do repetative testing of settings. No need to spend more time on this.

vonnieda, 2017-01-29:
@dzach I think maybe I am just not understanding what the "pick and place in place" does. I'm confused by your mention of having to create fake feeders. Can you just explain how you'd use this?

Specifically, where does this button go? (Feeder tab, Pick and Place tab?)
And what does it do? Does it just pick a part from the feeder and then put it right back?

dzach, 2017-01-29:
Its purpose is to help in repetitive testing of machine settings, so its appearance on panels that relate to production may not be useful at all. Also, if one wants to test a nozzle-tip or set the delays for a pump/solenoid valve, then it may not really matter if the part comes from a feeder or if it lies on the bed. 

Maybe this can be done with some javascript code. But as I said, if it doesn't ring a bell, then let's not spend more time on it.

vonnieda, 2017-01-29:
@dzach I'm not suggesting we don't do it. I just don't really know what "it" is :) Just asking you to help me understand exactly what *you* would use this for, so I can understand the use case. 

If you want to skip it, that's fine, but I'd prefer if you tell me what it does. What I am looking for is basically a description of exactly how you'd use it.

For example:
* I would put a part on the bed, position the nozzle over it and then click the button. It would pick up the part and then put it back down. 
* Or, I would select a feeder and click the button. It would pick the part and and then immediately put it back down at the spot spot.

That's all. Just want to understand the use case so I can see if it would be a good feature to add.

dzach, 2017-01-29:
Yes, what you describe above, is exactly what I did earlier to test my settings. Just didn't want to reprogram a job every time I changed the part or the location for the test. Separate 'pick' and 'place' buttons would do that for the feeder panel, but would not do it for the Pick and place panel, where the pick and place locations differ. So I 'm not certain where the button should go.

There is a number of actions that are only required  during the configuration of the machine, and rarely after that. Maybe a good place for such a button would then be on the Machine Setup tree or panels, possibly in the nozzle, or nozzle tip configuration panels.

vonnieda, 2017-01-29:
Yea, I think these would probably be better suited in the special commands section, or in the machine setup section. Or maybe even just as a script as I don't think they are general purpose enough to be included in the main UI.

I've updated the first comment in this issue with the changes that I think we should make. If you disagree with this can you let me know what you'd like to add?

dzach, 2017-01-29:
I think  a script will be fine. My ultimate wish would be an easily accessible  panel where one could create icons attached to scripts, "macros" so to speak. I have only experimented briefly with some of the canned scripts, and am not fully aware of the capabilities of the script API. Will visit it again sooner than later, I think.

vonnieda, 2017-01-29:
I like that idea. We could have a Macros subdirectory of the scripts directory and anything in there becomes a button in a Macros panel.

tracklogic, 2017-02-23:
I've been thinking about this earlier - for testing purposes - i'll want say the loose part feeder to be able to be picked from, then i can jab the "test alignment" button, and then be able to return the part back to where it was picked from, i've a feeling (although not tested yet), currently it will pick the part, test the alignment, but then it will drop it into the discard bin? It doesn't feel like it will take much to implement this, so i'll have a go (selfishly its going to be very useful for me in the next couple of days).

I think though, its going to have to be per-feeder type - reason being - an auto feeder, you won't be able to place the component back as the tape will of already moved forward, unless we have a mechanism where we can pick from an auto feeder but not advance it?

BendRocks, 2017-04-02:
>> but i think its easy - we call pick but no feed, so the tape won't move forward, so we should(!) be able to drop the component back into the tiny tape pocket.

The Yamaha feeders use the application of air to "cock" the feeder, and once you turn the air off, the feed happens. IOW, once you apply air to open the little cover, there is no turning back. And once the cover has slid back into place, dropping the nozzle down isn't good as it hits the metal cover. 

Your description is much needed for sure.  I think a dlg after the test alignment that that minimally offers "return to pocket" or "dump" would be ideal for me...Even better might be two more buttons that offer "return to pocket and re-pick & align" and "dump and repick & align".