openpnp/openpnp #435: Add vacuum sense button to Nozzle tip config tab

la3pna, 2017-01-29:
On the nozzle tip config page, there would be great if there was a button that request the vaccum level, to make it easier to do setup for the nozzles.

la3pna, 2017-01-29:
Should add that I'm not sure if this should just request sensor level, or if it should turn on vacuum pump and valves, then request vacuum level. 
I'm leaning against the first, but it requires a bit more thougth during setup.

vonnieda, 2017-01-29:
As noted in IRC, this is a bit of a complex issue. Currently the vacuum sensing is a feature of GcodeDriver. Putting it there was a good way to get this feature in quickly without making major changes through the system, but before we can do things like expose that data in the main UI we need to move it into the SPI interfaces.

This is likely going to require a similar setup as we have for Bottom Vision and Fiducial Checking. We'll need a PartDetection interface or something that allows us to run the checks in an implementation agnostic fashion.

References:
* Pick Fail Detect and Discard #102

vonnieda, 2017-02-12:
With the changes made in https://github.com/openpnp/openpnp/blob/develop/CHANGES.md#2017-02-12, I think this is no longer required since you can use the new Actuator UI to quickly read the value. @la3pna, if you disagree and think we need a dedicated button in nozzle tip, reopen this.

closed 2017-02-12 by vonnieda

la3pna, 2017-02-12:
> openpnp/blob/develop/CHANGES.md#2017-02-12, I think this is no longer

This solves it in a better way than my proposed idea, so no disagreeing
from me.

Thomas.