openpnp/openpnp #439: Should Machine Setup tree start up expanded?

dzach, 2017-02-01:
I just deleted most of the example feeders and all of my test feeders in the `Machine Setup` tree,  because I got tired of collapsing it in order to make space for the setup panels at the bottom of the right pane.

Wouldn't it be nice to have a checkbox that expanded / collapsed the tree according to the situation in hand? Like this perhaps?

[image]

or, collapsed to the first level:

[image]

If you like the idea, I have the code ready to push.

Cheers!
Dimitrios

vonnieda, 2017-02-01:
Looks good to me, although just so I know we're on the same page... you shouldn't have to collapse it to see the setup panels. You can scroll it. This is how mine looks normally:

[image]

dzach, 2017-02-01:
Ehm, I meant I don't want to keep enlarging and shrinking the top and the bottom panes to find what I need in each one of them. What I'd like is to have one branch of the tree open, the one I'm working on, but not have to go through all the leaves and collapse them manually. Having all expanded at start up is overwhelming.

vonnieda, 2017-02-01:
Fair enough, PR when ready :)

vonnieda, 2017-02-02:
Fixed in #440, thank you for the nice improvement @dzach!

closed 2017-02-02 by vonnieda