openpnp/openpnp #443: MJPEG IP Camera streaming support

PeeJay, 2017-02-04:
A while ago I implemented basic IP camera support in the OpenCVCamera class. However, it was an absolute nightmare getting OpenCV to work correctly as it required recompiling with ffmpeg support, but it never did work very reliably.
In the last few days I decided to try again, and discovered there is a new OpenPnP driver based on the excellent webcam-capture library. With only minimal code I got it running perfectly with two streams at once.
The code can be found at https://github.com/PeeJay/openpnp/commit/e1050884f7e9637a00a9c79ad0952cd8a768ab0f
It's not ready for merging yet, as I need to find a convenient way to add/remove cameras.
I just thought I'd open an issue as a placeholder in case anyone else is looking for something similar.

vonnieda, 2017-02-04:
@PeeJay Can you explain a little more about what this does? I'm not quite clear on what the composite camera is for. A brief rundown of what this is and how it works would be helpful.

PeeJay, 2017-02-05:
From what I can gather ( I copied that bit from an example) the default driver is the WebcamDefaultDriver which is used for USB cameras. To use IP cameras the IpCamDriver needs to be loaded, but to use both drivers at the same time (ie IP and USB) then a composite driver class is needed to load both drivers. Perhaps a better name would be IPandUSBdualDriver (or similar) rather than composite, which reminds me of composite video.

The two calls to IpCamDeviceRegistry.register add IP camera devices to the list that is returned by Webcam.getWebcams(), so they can be just selected and used the same as a USB camera.
[image]

vonnieda, 2017-02-05:
Thanks @PeeJay, that makes sense. So it seems we may need to add a new camera type in OpenPnP like IpCamera, and this would use WebcamCapture under the hood and allow you to specify a URL.

cri-s, 2017-02-12:
Yes, normally cameras are defined inside XML, i include one example below.
Here is one example of combined driver,
https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/java/DualNativeAndMjpegWebcamExample.java
I don't have included that, because it exist serious bug that is already resolved, but not included inside maven distribution. 

<?xml version="1.0" encoding="UTF-8" ?> <storage> <ipcam name="Dasding 01" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=1" /> <ipcam name="Dasding 02" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=2" /> <ipcam name="Dasding 04" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=4" /> <ipcam name="Dasding 06" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=6" /> <ipcam name="Dasding 07" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=7" /> <ipcam name="Dasding 10" url="http://www.dasding.de/ext/webcam/webcam770.php?cam=10" /> <ipcam name="Biblione Palace" url="http://webcam.hotelbibionepalace.it/mjpg/video.mjpg" mode="push" /> <ipcam name="Foscam" url="http://50.197.211.181:8910/videostream.cgi" mode="push"> <auth user="user" password="foscam" /> </ipcam> </storage>

cri-s, 2017-02-12:
XML was not displayed, here one example https://github.com/sarxos/webcam-capture/blob/master/webcam-capture-drivers/driver-ipcam/src/examples/resources/movable.xml

vonnieda, 2017-04-13:
PR will be accepted if sent, no need for this issue.

closed 2017-04-13 by vonnieda

dzach, 2017-04-17:
@PeeJay 's MJPEG branch is the only way I have found so far to work, to set my Philips SPZ5000 USB webcams to 1280x1024 px, using `mjpg_streamer` as an IP streamer. All other drivers, OpenCv, webcam fail to put the camera to the 1280x1024 `MJPG` mode and default to the native 640x480.

PeeJay, 2017-04-18:
@dzach  I was originaly planning to tidy it up and submit a pull request, but the way the webcam-capture library works makes it rather difficult to do well.  Instead I decided to wait until the new openpnp-capture library is sufficiently implemented and add support there.

dzach, 2017-04-18:
@PeeJay I have merged your MJPEG branch into my local working one, and am happy with the result. I can now manipulate the webcams with `mjpg_streamer` which isn't too picky with them. The webcams can yield 1280x1024 each @80% JPG compression, and I have them both happily working on a USB 2 hub, together with the motion controller. 

I tried including a text field to the MjpegIPCamera wizard to be able to enter/change the URL, but didn't go too far as it is more involved than I initially thought. I still have the URL's hardcoded in the source file. 

One problem I have with openpnp is that it takes more than 2 core's load on this 4 core CPU , often more than 60% of total CPU capacity, for 2x 1280x1024 cams. Firefox, displaying the same mjpg_streamer streams at the same time, is less than 25% of CPU load, while mjpg_streamer is at less than 1% of the total CPU capacity.

[image]

So something is very inefficient in this setup. I noticed that OpenPnP is continuously writing the incoming images to disk, in /tmp, in Linux. Taking into consideration that file events may be monitored by many processes, maybe this explains why it is so heavy on system resources. Limiting FPS on the camera displays doesn't seem to be doing any difference.

In any case, thank you for your driver. It's a big help for me.