openpnp/openpnp #444: NamedCsvImporter Needs Work

vonnieda, 2017-02-07:


cri-s, 2017-02-12:
simplified greatly with the following ideas in mind:
Delimiters are automatically detected.
Non matching lines are automatically ignored.
It imports CSV files, not Excel files. 
The second file is Exel file, it have excel ""delimiter"" instead of CSV "delimiter" .
Some limits was chosen, format specs need to be within the first 10 lines, and it need to be present 6 valid fields in order file is imported. In this file format specs is on the eleven line. Its possible to change this, changing this for, its increases wrong import risk too.
for (int i = 0; i++ < 10 && (line = reader.readLine()) != null;) {

vonnieda, 2017-03-16:
Note that work done in #461 should resolve the Altium issue.

vonnieda, 2017-04-13:
Recent changes have improved this enough. If additional changes are needed, new issues can be opened.

closed 2017-04-13 by vonnieda