openpnp/openpnp #445: Distorted graphics

dzach, 2017-02-07:
Work in #421 has revealed a problem with distortion of graphic elements in OpenPnP. Here are some examples from Linux, Kubuntu 16.04 LTS, 64 bit:

[image] OpenPnP icons

[image] Right icon above, rendered by Inkscape

[image] Similar icon rendered by Firefox

[image] OpenPnP camera reticle

Distortions were first identified in SVG icons, but are present in other OpenPnP graphics too, as the reticle example above shows.

Are these OS specific? And if so, then to what extent? Are they due to some Java quirk that needs to be properly set, or are they a Java bug?

vonnieda, 2017-03-21:
This appears to be an issue with Java2D rendering of circles. An issue has been filed with Batik about it, but I think it's likely that it's lower level than that. For reference, the issue is https://issues.apache.org/jira/browse/BATIK-1175and has seen no activity in the 2 months since it was filed.

Given that this does not significantly affect functionality, I am going to close this. If anyone feels strongly that this needs further attention, please reopen it, but only if you intend to fix it.

closed 2017-03-21 by vonnieda

dzach, 2017-03-23:
Good to know where this problem comes from. At least it is not from the SVG creation tools we are using.