openpnp/openpnp #448: OpenPnP Camera Capture Library

vonnieda, 2017-02-16:
This ticket is a placeholder for a new capture library I am working on. This library is meant to add a native option for camera capture on Windows, Linux and Mac and will give us the ability to control cameras at a much lower level, solving several problems. The goals are:

1. Bundled support for Win 32/64, Linux 32/64, Mac 64.
2. Native camera access on each platform. Implemented as MediaFoundation, V4L2 and AVFoundation.
3. Named device enumeration.
4. Strong, repeatable unique IDs.
5. Codec control with support for at least YUV and MJPEG. MJPEG will allow the use of multiple USB cameras on a single hub.
6. Auto and manual exposure control.
7. Auto and manual focus control.
8. Stretch Goal: Zoom control.
9. Simple, common C wrapper for the listed APIs.

Work will be done in https://github.com/openpnp/openpnp-capture. The resulting library will be usable as a C library and is designed to be easily bindable, with the end goal being a Java binding with an all-included Maven package.

kakemoms, 2017-08-27:
The capture test program works with my 8MP chinese USB cam, but it has problems with the highest resolutions in MJPEG. Basically it lists all the IDs of the modes, but choosing anything above 2048 x 1536 just ends the program (see attachment - last part). For 2048 x 1536 it gives a slightly higher frame rate than what is given as supported (see attachment - first part).

[openpnp-test.txt](https://github.com/openpnp/openpnp/files/1254626/openpnp-test.txt)

vonnieda, 2017-10-01:
This is now complete and released.

closed 2017-10-01 by vonnieda