openpnp/openpnp #449: Importing CSV data

PaulDAR, 2017-02-20:
Hi,  Newbie question.
Have built a replica of Jasons reference design, with Betz Technik nozzle adapters and Robotdigg vacuum pump being the only improvements, Implemented after reading about here. Thanks. Everything ran fine for Setup and Calibration.
There was just one issue with smoothie board after loading config file. On power-up of smoothie board, There were ‘no match in class’ errors for the following 2 lines, which I then just deleted.
      <job-planner class="org.openpnp.planner.SimpleJobPlanner"/>
      <job-processor class="org.openpnp.machine.reference.ReferenceJobProcessor" demo-mode="false"/>
Just highlighting above and wondering if operation is affected. ( I come from hardware background)
When selecting ‘File’ ’New job’ in Openpnp on start-up, No dialog box appears. Unless there is already a board already loaded, In which case there is a prompt to save. Is its sole purpose to remind user to save current job, because there is no follow up dialog to load new job?
What I would like to do is load data from a CSV file taken from Excel. So my question is if there is any more information/examples available than that currently on ‘Importing Centroid Data’ page (Am aware of issue #444) ?
Many thanks for any help.
Paul

PaulDAR, 2017-02-20:
Hi 
Just noticed 2 copied Lines did’nt appear, guess I’m not aware of some protocol, Sorry.
The lines appear at line 76 and 77 when using Sublime test editor.
 Line 76 beginning ---job-planner class =….    
 Line 77 beginning --  job-processor class =….
Thanks
Paul

vonnieda, 2017-02-20:
Hi @PaulDAR,

New job just creates a new, empty job. When you have a job loaded that has had changes it prompts because that job is going to get unloaded a new one will be created. 

To import job data just to the main menu -> File -> Import Board and the various options will be listed there. The normal process is:

1. Create a new Job
2. Create a new Board by clicking the green + under Boards
3. Go to File -> Import Board to import data into the new board

As far as loading data from Excel, the Named CSV Importer should work okay for you. Just make sure to model your data after the example shown at https://github.com/openpnp/openpnp/wiki/Importing-Centroid-Data#others. Alternately you can use the Eagle mountsmd.ulp importer and just export your file to be similar to what it expects. The format is much simpler.

You can see some samples for that at https://github.com/openpnp/openpnp/blob/develop/samples/Demo%20Board/Demo%20Board%20v2.mnt and https://github.com/openpnp/openpnp/blob/develop/samples/Demo%20Board/Demo%20Board%20v2.mnb. MNT is the top of the board, MNB is the bottom.

PaulDAR, 2017-02-20:
Hi Jason

Many thanks for your detailed reply,

Much appreciated

Paul D'Arcy

closed 2017-02-20 by vonnieda