openpnp/openpnp #454: Ability to have special characters in GcodeDriver commands

vonnieda, 2017-03-01:
Currently it's not possible to have non-printable characters in the GcodeDriver commands. This has been requested by two users who use this in a customized version of Grbl. The issue seems to be twofold:

* It's not possible to add the special characters in the UI.
* Even if you add them manually in the XML the parser blows up.

We should add a way to add escape codes to the commands like \uXXXX and have these translated at a low level in the GcodeDriver.

More discussion at https://groups.google.com/forum/#!msg/openpnp/h7LA-vQrzGA/IYJ9zi2GAwAJ

cri-s, 2017-03-06:
It is thinkable that substitutions works, example (-CTRL-X-) or whatever template get substitued to 3 sequences of ctrl-x ´.
This substitution need to work on connect/disconnect and homing .

vonnieda, 2017-04-13:
Closing this as too machine specific. This isn't valid Gcode and I don't think we should add it. Better to fix the motion controller.

closed 2017-04-13 by vonnieda