openpnp/openpnp #460: import board named CSV crash if CSV is opened in excel

Misfittech, 2017-03-13:
import board named CSV crash if CSV is opened in excel

Cannot create file "C:\Users\TSTERN\Google Drive\projects\Stepper_feedback\hardware\SmartStepper_NEMA23_A5995\Project Outputs for SmartStepper_NEMA23_A5995\Pick Place for SmartStepper_NEMA23_A5995.csv". The process cannot access the file because it is being used by another process

Exception Occurred In  GenerateReport

Misfittech, 2017-03-13:
Might not have been an openpnp issue

closed 2017-03-13 by Misfittech

cri-s, 2017-03-17:
This is because excel uses ""foo""  instead of "foo" .
Never less I have seen some very wrong CSV from excel, where refid have one opening " without closing " and the rest have double " for opening and closing. 
If your  need it reopen the issue and I add it.