openpnp/openpnp #462: bsh is not working on windows build fc9736

cri-s, 2017-03-15:
I'm open this low priority issue for someone other.
bsh scripts don't appear on script menues, including after new scan, so i assume that bsh is not recognized
as valid scripting resource. As i have distributed various .bsh script to some users receiving feedback
, i know that in some cases it worked. I include a bsh script that center camera over fiducial using hough circles.
[centerCircle.zip](https://github.com/openpnp/openpnp/files/844252/centerCircle.zip)

Regards
Cri.

vonnieda, 2017-03-15:
Confirmed not working on Windows. Works okay on Mac. Choosing Menu -> Scripts -> Examples -> Print_Scripting_Info.js shows that the bsh interpreter is not being loaded. Not sure why. Could be a Windows specific bug in bsh.

```
2017-03-15 08:44:23 SystemLogger INFO: Scripting Environment
2017-03-15 08:44:23 SystemLogger INFO:   Global Variables
2017-03-15 08:44:23 SystemLogger INFO:     config: org.openpnp.model.Configuration@7a664378
2017-03-15 08:44:23 SystemLogger INFO:     machine: org.openpnp.machine.reference.ReferenceMachine@59eee832
2017-03-15 08:44:23 SystemLogger INFO:     gui: org.openpnp.gui.MainFrame[frame0,77,41,1523x914,invalid,layout=java.awt.BorderLayout,title=OpenPnP - Untitled.job.xml,resizable,normal,defaultCloseOperation=DO_NOTHING_ON_CLOSE,rootPane=javax.swing.JRootPane[,8,31,1507x875,invalid,layout=javax.swing.JRootPane$RootLayout,alignmentX=0.0,alignmentY=0.0,border=,flags=16777673,maximumSize=,minimumSize=,preferredSize=],rootPaneCheckingEnabled=true]
2017-03-15 08:44:23 SystemLogger INFO:     scripting: org.openpnp.Scripting@2c93e75f
2017-03-15 08:44:23 SystemLogger INFO:   Scripts Directory: C:\Users\Jason\.openpnp\scripts
2017-03-15 08:44:23 SystemLogger INFO:   Scripting Engines
2017-03-15 08:44:23 SystemLogger INFO:     Engine: Oracle Nashorn (1.8.0_66)
2017-03-15 08:44:23 SystemLogger INFO:       Language: ECMAScript (ECMA - 262 Edition 5.1)
2017-03-15 08:44:23 SystemLogger INFO:       Extensions: [js]
2017-03-15 08:44:23 SystemLogger INFO:     Engine: jython (2.7.3)
2017-03-15 08:44:23 SystemLogger INFO:       Language: python (2.7)
2017-03-15 08:44:23 SystemLogger INFO:       Extensions: [py]
2017-03-15 08:44:23 SystemLogger INFO: 
```

vonnieda, 2017-03-24:
This may be related https://github.com/beanshell/beanshell/issues/24

vonnieda, 2017-04-03:
Fixed in https://github.com/openpnp/openpnp/commit/06b337d282124db35689f7d7b668658ef98431a4

closed 2017-04-03 by vonnieda