openpnp/openpnp #466: Feeder configuration not always saved/apply

Misfittech, 2017-03-17:
With the development branch as of 3/17/2017 when creating a strip feeder and setting the location using the camera and then changing the Z height. The software will not use the Z height unless OpenPnP is closed and reopened.

Misfittech, 2017-03-17:
Actually it might be that if the Z height is set and then you reset the location using camera internally the Z height is set to zero but the GUI still shows old height. After reboot the Z height is zero and can be changed

vonnieda, 2017-03-18:
It's possible there is a property notifier absent. Can you try to track down the exact process that causes the error and report those steps please?

Misfittech, 2017-03-18:
I have verified what seems to cause this problem.  If you go to strip feeder and select the starting hole and second hole with camera on a previously configured strip feeder (where Z is no zero) you will notice the X and Y change but the Z remains.  Now if you do not hit apply but instead try to run machine it will have set the Z to zero internally and run with Z of zero.  If you go back to screen and hit apply it will then  apply the Z value and run correctly.

vonnieda, 2017-03-18:
I am not able to reproduce this. Can you please record a short video showing the problem, or write steps by strip instructions to reproduce? For reference, here is what I tested:
1. Set up a strip feeder with two holes at Z -10 and -11. Hit Apply.
2. Make sure camera is at Z = 0 using Z Park button.
3. Move to a new position near first hole and capture location using square blue camera button. Do not apply. X and Y update, Z does not. (As expected)
4. Move to a new position near second hole and capture location. Do not apply. X and Y update, Z does not. (As expected)
5. Perform a Feed and Pick using the Feed and Pick button. Note that nozzle goes to Z -10 as expected.
6. Click Apply, no fields change a expected.
7. Perform a Feed and Pick. Note that nozzle goes to Z -10 as expected.

This all appears to be behaving as designed.

vonnieda, 2017-03-21:
Closing as not reproducible. If this can be shown reproducible please reopen and post instructions to reproduce.

closed 2017-03-21 by vonnieda