openpnp/openpnp #467: GUI non reponsive

Misfittech, 2017-03-17:
With the development branch as of 3/17/2017  When running a Job the GUI becomes no reponsive. The job will run to completion at normal operating speed but the GUI will never come back and will have to killed externally.

vonnieda, 2017-03-18:
Can you provide instructions to reproduce please? In particular:
* What OS and version?
* machine.xml, job.xml, board.xml.
* Is the UI locked up during the entire run, or does it just fail at the
very end?

Misfittech, 2017-03-18:
[openpnp.zip](https://github.com/openpnp/openpnp/files/852372/openpnp.zip)

What seems to repeat GUI unresponsive is to got to a reference strip feeder that is already configured and then try to set the hole locations again.  Here is a video of the GUI locking up running binary download, but have seen the same behaviour running source code. 
[ice_video_20170318-074421.zip](https://github.com/openpnp/openpnp/files/852374/ice_video_20170318-074421.zip)

Misfittech, 2017-03-18:
Here is a video showing it locked up. During the video the pick and place operation is running at normal speed the image processing on up camera and down camera for strip feeders is working. 

This run was done from the source code running from eclipse, but it happens on the binary build as well. 

[openpnp_lockedUp.zip](https://github.com/openpnp/openpnp/files/852407/openpnp_lockedUp.zip)

vonnieda, 2017-03-18:
Some things to check:
1. It looks like you are running this remotely, maybe? With Team Viewer? Please run it locally without Team Viewer. That adds a whole layer of complexity and too many variables to debug.
2. I did not see evidence of a lock up in the first video (ice_video_20170318-074421.zip). Was this a mistaken upload? It looked okay and appeared to exit normally.
3. What OS and OS version are you running? And how much RAM is available?
4. When running from source code, make sure you are setting max memory to something > 1GB. This setting is cooked into the binary installer. For reference, my Run Configuration is below. The important part is the -Xmx1024m.

[image]

trampas, 2017-03-19:
I was not using teamviewer. 

The OS is windows 10 x64, on a thinkpad T61 T7700 CPU and 8GB ram. 

The lock up is the the GUI interface becomes non responsive,  while the job and image processing continues to run correctly. It is like the GUI gets stuck waiting on file, camera img, etc which never comes available.  The CPU usage is not high when this happens so again I thinking stuck on a resource? 

Trampas

vonnieda, 2017-03-19:
@trampas Do I remember you saying you switched from OpenCvCamera to WebcamCamera? If so, was there a specific reason for that? I recommend people use OpenCvCamera unless there is a very good reason not to, as WebcamCamera is much less tested. This could be one thing that is different from your setup and others. Most people use OpenCvCamera.

trampas, 2017-03-19:
I have head as opencv and up as webcam. The opencv kept causing problems
when starting OpenPnP. On startup sometimes the cameras would not show
video.  I was having to restart OpenPnP several times to get it to show
camera. Sometimes it would then mess up and swap cameras too.

vonnieda, 2017-03-19:
Okay, can you try using just one or the other? It could be that they are conflicting. In general, if you have two basic USB cameras the OpenCvCamera should work fine for you, as long as they are on different host controllers. There is more information about that here: https://github.com/openpnp/openpnp/wiki/USB-Camera-Troubleshooting-FAQ.

If this were my machine, and I was having this problem, here is what I would do:

1. Backup my machine.xml so I don't lose anything.
2. Make sure each camera is on it's own host controller. This usually means different ports at the computer, even if there is a hub between the camera and the port. See https://github.com/openpnp/openpnp/wiki/USB-Camera-Troubleshooting-FAQ for details.
3. Remove the Webcam camera from OpenPnP and restart it.
4. Add the camera back as OpenCvCamera.
5. Run further tests and see if you still experience lockups.

Some things to be aware of:
* When OpenCV enumerates USB cameras it does it based on port address. As long as you keep the same cameras plugged into the same ports, they will come up in the same order.
* If you find that you have difficulty connecting more than one camera at once with the OpenCvCamera, that is due to the "USB troubleshooting" stuff I linked above. It all comes down to having each on their own host controller.
* If your machine does not have two USB host controllers then you have to look for alternate solutions, such as the WebcamCamera interface. But, like I said, this code is much less tested and less trusted. It's harder for me to diagnose, as well, as I have less experience with it.

I definitely recommend focusing on fixing the lockups first, and then worry about further configuration. OpenPnP should not lock up regularly. I run it for hours and hours at a time on my machine and so do many other people. Windows 10 is a well supported OS. So, we just need to track down what is different about your setup that is causing this issue, get it fixed and move on from there.

One other thing to note, which may help alleviate some of the pain: If OpenPnP locks up and you have to hard kill it, any config changes you made since starting it will be lost. This may account for some of the oddness you are seeing in the other tickets you have filed. This is why I suggest getting the lockup fixed first and then worry about the rest.

tracklogic, 2017-03-20:
Hi not sure if this is related, but i've noticed it stays in the "clean up" stage for a long time on my machine with the new logging u/i changes and the log file in debug mode.

The more jobs you run, the longer it is in the "clean up" state. Resetting the log file before the job starts seems to help.

Might be completely unrelated though - i think somethings intensive in the new logging U/I but not had chance to debug.

vonnieda, 2017-03-20:
@tracklogic When you say "resetting the log file", are you talking about clearing the OpenPnP.log or using the clear (red x) button in the log panel?

vonnieda, 2017-03-20:
Okay, I just did a quick test running the pnp-test job with TRACE logging on and it did show some pretty serious UI churn. Eventually the UI pretty much stopped updating, or kind of went into freeze frame mode. I'm going to see if I can revert the logging feature for now until we get this figured out.

vonnieda, 2017-03-20:
Having tested further, this pretty closely replicates the behavior that @Misfittech reported. Also, my apologies, I seem to have mixed up @Misfittech and @trampas earlier in the thread.

In any case, I just ran the pnp-test job fully through and about 1/3 of the way through the UI stopped updating completely while the job continued to work in the background. During this time the UI was unresponsive to input, as well.

After reverting the logging changes (git revert 5eb5c8792841e636ad0f89f02a60efb2c33fbe69 -n -m 1) this problem goes away. So, I suspect we have some kind of UI issue with the new logging changes. I should note that I did not have a search filter applied in the log panel and I had logging set to TRACE.

For now I am reverting the logging changes and will comment further on #288 regarding how I think we can fix this.

vonnieda, 2017-03-20:
@Misfittech and @trampas, can you please download the newest version and see if this problem is resolved?

trampas, 2017-03-20:
@misfittech and @trampas are the same person... so no mixup.  I will
download and give a try.

vonnieda, 2017-03-20:
@trampas Ah, that explains it then! :)

vonnieda, 2017-03-21:
Users have confirmed this is fixed with the reversal of the logging changes.

closed 2017-03-21 by vonnieda