openpnp/openpnp #469: Long strip feeders not working after ~50 parts

Misfittech, 2017-03-18:
I have noticed that after ~50 parts the strip feeder does not work.  From the camera view it appears that OpenCV is predicting the location based on the first two pick points and after a large number of parts this prediction error has increased to point it can not compenstate with vision.  

Would it be better for user to pick the starting hole/part, second hole/part for spacing and then the last hole/part on the strip for number and better interpolation inbetween? 

Trampas

Misfittech, 2017-03-18:
A releated idea which might be easier is when configuring strip feeders OpenPnP could setup as done today but during the setup move to each hole with camera, calculate the pick location, and continue to end of strip calculating each part location.  
Overall this would make the job run faster.   Specifically the time at start up to move the camera 4mm is lot less than time for each part to move between camera location and nozzle location (about 60mm for my machine).  Hence at configuration finding each part pick locaiton and storing would be fast.

vonnieda, 2017-03-18:
* Do you have "Use Vision?" turned on?
* Did this happen after a fresh startup, or did it happen after having previously picked from that feeder during the same run? In other words, was it the first pick from this feeder since starting OpenPnP?

We store (in memory) the last point captured by the vision system and that is used for each further calculation of the line to follow, but we don't store that in the XML, so I can envision a scenario where this would happen:
1. You configure the feeder and run it through 45 parts.
2. Each time you feed there is a bit of drift, but the last saved hole position corrects it.
3. You restart OpenPnP.
4. There is now no previous vision location, so we can only calculate the line based on the first two holes.
5. We go to the expected position for the 50th part and we're too far off to find the hole.

I think we can fix this by just storing the lastHoldLocation variable in the XML between runs so that OpenPnP can find that hole again after a restart.

Note to @Misfittech: OpenPnP does allow you to use the first and last holes in a strip rather than the first and second, and this does cause it to be much more accurate over the length of the strip. You can try that as a temporary measure.

anthonywebb, 2017-03-18:
I'm having deja vu from this post. A ways back there is a post from me that reads almost exactly the same. 

We use this type of feeder almost exclusively and have learned to work around it. What we have found is that on paper tape the holes and part pockets will vary and by sometimes quite a lot. 

Our solution has been to anchor one end of the tape and stretch the tape manually until the last part is where we want it to be based on our part to part measurement expectations. Yes, it is a total PIA, and it is a huge time sink for us which is why we are gunning for auto feeder options to start to surface soon. 

We have tried to enable vision hoping that would help but we have not had luck with it. I'm not sure why, and we have not taken the time to dial in the opencv pipeline, add really great lighting, etc.

What we need to do is break away from what is working for us long enough to be able to diagnose why vision is not helping find the proper part, but we always seem to be under the gun to be putting out boards that we always revert to what we know works just to get the job done. I'll try and carve out some more time to help get Jason the info he needs to improve this, and, like I say, it may just be an issue in our camera setup/config.

cri-s, 2017-03-19:
I could share a strip feeder,
 simple first, next, last feeder with optional vision using either templatematch or reflection from the two pins like resistors, cap, ... 
No holes vision.

trampas, 2017-03-19:
Jason,  Yes I reboot OpenPnP **a lot** as it locks up **a lot**. I do use vision on strip feeders. 
I might try using the first and last hole to see if it helps. 

Anthony, thanks I was thinking about doing a template match in OpenCV to find part pick location which is simple in OpenCV, but learning Java enough to do it in OpenPnP is tough for me.   The basic idea is when configuring strip draw a box around the part and select the pick point.  Then after the holes are found and it before picking part the software would do a template match on the part to fine tune pick point.  There is more to this than it sounds as you need to do some clean up of glare from part, etc.  

Either way I have to make some personal decisions:

1. Learn Java and OpenPnP enough to contribute 
2. Kill my PnP  project all together and buy something (with support)
3. Write my a script in Python to run machine. 

I would like to the first one so others benefit from work,  but at the point I am scared of the un-quantifiable time it will take to learn Java and OpenPnP.

vonnieda, 2017-03-19:
@trampas I suggest addressing the root of the problem first. OpenPnP does not commonly lock up as far as I know. I've posted some thoughts on your other ticket regarding that: https://github.com/openpnp/openpnp/issues/467. If you can respond to that we can track down that issue, get it fixed and then you will have less difficulty making progress.