openpnp/openpnp #471: Confirm changes when changing machine setup

trampas, 2017-03-21:
When clicking the buttons (blue camera or blue nozzle) buttons to set locations, like the camera location. If the location has previously been set, ie non zero. Then there should be a pop up asking if you to confirm you want to change settings. 

This will help as confusion over the set icons verse move icons happen.

vonnieda, 2017-03-21:
Similar to #151. I think a hybrid approach would be ideal:
* By default, pop an "Are you sure?" dialog when clicking one of these buttons and the value is not zero.
* Hold down a modifier key to skip the dialog so that users have a reasonable way to skip them.

vonnieda, 2017-03-24:
Played around with this a bit, patch below. Adds the functionality described above to the LocationButtonsPanel. It would also need to be added manually to places that don't use that widget, like in BoardLocation and Placement editors.

[patch.diff.txt](https://github.com/openpnp/openpnp/files/866649/patch.diff.txt)