openpnp/openpnp #473: Apply button

trampas, 2017-03-21:
Some pages, like the feeder have an apply button. While other pages like parts tab does not. This creates confusion on operator as to when they need to click apply or not as the software is not consistent.

vonnieda, 2017-03-21:
Similar to #151. I think a hybrid approach would be ideal:
* By default, pop an "Are you sure?" dialog when clicking one of these buttons and the value is not zero.
* Hold down a modifier key to skip the dialog so that users have a reasonable way to skip them.

vonnieda, 2017-03-21:
I agree. I've been thinking about this quite a bit over the last few months and I've semi-decided to just remove the Apply and Reset buttons entirely. I don't think they add much to the experience.