openpnp/openpnp #474: Board Alignment

Misfittech, 2017-03-22:
When doing board alignment the function should return the maximum alignment error and report this to user, for example a pop up dialog box saying "Maximum error is  XXmm, if not acceptable try aligning board again."

Additionally when doing an offset and rotation alignment with two points the offset should be calculated as the mean offset of the two points to reduce error.

Misfittech, 2017-03-22:
I should mention that currently it appears that the code does not scale the axis based on the two pick points. That is if we have two parts one at (0,0) and one at (1,1) and when we locate these points using camera as locations (100,100), and (102, 102) then the board offset would be 100, 100, and rotation angle of zero. However based on the second point when apply the offset and rotation the part location would be (101,101) which based on the picked point is a 1mm error in each axis.  

If we averaged the pick points for the offset then in the above case the offset would be (100.5,100.5) and the resulting error is +/-0.5mm. In this case the error is distributed between the two points. 

Ideally OpenPnP would allow more than two points for alignment and do a least square fit. Even better would be to have a user option which would allow and Affine transformation.  This would help correct for machine errors and or board errors.  

Once multiple part alignment is done the next step would be to use the solder paste gerber layer to get template around part. The square size could be dynamically determined by counting number of SIFT features. Then using the  template a match could be done with the camera image to find the location of the part.

cri-s, 2017-03-23:
If fiducials are out more then 0.4mm , you probably could throw away your stencil, adjust placement positions visually and test it using double sided tape, update stencil data and reorder it.
You see it on stencil setup, or when you check the stencil against PCB.  In case you require 0.5 or 0.3% max boul and twist,  you can adjust the constant inside fiduciallocator for your branch.
The actual value is a good compromise between IPC limit and pbc producers limits specs. I propose to close this issue.

Misfittech, 2017-03-23:
So what I have found is that when I try to align a board using the two part alignment.  I was unable to get my alignment of the board, relative to the Altium pick and place files to be better than 0.5mm.  
First thing I noticed is that after two part alignment the first part was correct, but second part was more than 0.5mm off.   After going back and doing the steps per mm setup again I can get the two parts pretty close, however other parts are still off.  For all know the problem could be the Altium file is wrong. 

Again based on the two pick parts, it would be helpful to the user to report the error after translation and rotation, for example if the dialog said "Alignment error is 0.5mm, if this is too high, try again or check your axis calibration (htttp://....)" 

Right now I can go and manually set the place locations for parts that are off to work around this problem until I can figure out what is the root cause.

vonnieda, 2017-03-23:
I highly recommend checking all assumptions. Get a pair of calipers and measure the board, and the distance between placements. See if it matches the file. Measure your measuring equipment, too. I once wasted a week of work because I had a steel mm ruler that I found out was .5mm off at 300mm.

Misfittech, 2017-03-23:
The basic idea of this issue/request is that from 2 points you can check the X and Y steps per mm. If you did three points it would be possible for the software to check the squareness as well.   I will look at the code and see if I can figure out how to do this. 
I personally think it would be a great check for users. That is if user picked three points on board and their machine was off it could flag it and let them know what to do to address issue. 

As far as my machine and problems go... 
I have ordered some aluminum plates to redesign parts of the machine. The plastic NEMA17 motor mounts are bowing from belt tension, and need fixed.  I have also noticed the L bracket for the NEMA8 mounts flex more than I like.   With that said I can very repeatably hit points and have gone back through the calibration of x and y again, but need to check squareness again which could be my issue as I have not checked in awhile, and I know that the motors have railed and could have jumped steps (dual belts on Y) which would change the squareness.

cri-s, 2017-03-23:
You assume that PCB boards are exact. That is wrong.
As example the optical calibration on PCB board was a total fail.

Misfittech, 2017-03-23:
It is back to assumptions....

You can assume machine is correct or PCBs, either way it is good to report
the difference.

Currently the alignment system assumes both are correct.

jkhax0r, 2017-04-02:
I recently had a run of relatively large pcbs (7"x4") with fiducials on the edges only.  It seemed that on a couple of the boards I did (maybe 2 out of 7), that the components towards the right were noticable all shifted.  I have calibrated my x / y with a ruler and run a # of smaller boards and never noticed this.

I came here specifically trying to understand how the 2 fid check was working since I've always been trained to use at least 3 and only seen machines that do it.  

I'll be running these board again in a month or so and would love to give the 3 fid with scaling a shot.   If you guys want to give me a few pointers I've looked at the code and it seems pretty straightforward and I'll take a stab.  Think I'll just need to understand the math and then where is best to surgically inject some board scaling to all component coords.

vonnieda, 2017-04-02:
@jkhax0r You'll most likely have to make the changes in calculateBoardPlacementLocation https://github.com/openpnp/openpnp/blob/develop/src/main/java/org/openpnp/util/Utils2D.java#L89. There are several versions of that method so you'll need to go through the code and find the best way to handle updating them and will likely need to pass the scaling data.

cri-s, 2017-04-02:
If you want work on that, mail me, (phone.cri@gmail.com) .
There are some source of errors:
board position adjust using only one fiducial and not the center of
fiducial1 and fiducial2.
On bottomvision the offsets need to be rotation adjusted before applying it.
That code seems to be disappeared from openpnp.
Overall the errors are small, and if you use center board zero
position, other consideration apply too.
You need to have round fiducials that can be detected with houghCircle.
Further doing Templatematch sometime depending on lightning condition
affsets the fiducials a bit.  You need to be able to  run .bsh scripts
in order to do some checking and check the error source that you have.

tonyluken, 2025-01-14:
I believe everything in this issue has been addressed since [PR #975](https://github.com/openpnp/openpnp/pull/975) was merged. It allows two or more fiducials to be used. If two fiducials are used, a single scaling factor, a rotation, and a linear offset in x and y are used to correct placements. If three fiducials are used, a full 2D affine transform is computed and used to correct placements. And if more than three fiducials are used, a least squared error affine transform is computed and used for correcting placements.

closed 2025-01-14 by tonyluken